rdaly525/coreir #632: Missing `}` in json output after running cullgraph

leonardt, 2018-10-11:
[pe.json.txt](https://github.com/rdaly525/coreir/files/2467191/pe.json.txt)

```
$ coreir -i pe.json -o tmp.json -p cullgraph
```

produces the following `tmp.json`
```
{"top":"global.PE__ops_add_sub_and__or___UIntKind_16",
"namespaces":{
  "global":{
    "modules":{
      "PE__ops_add_sub_and__or___UIntKind_16":{
        "type":["Record",[
          ["I0",["Array",16,"BitIn"]],
          ["I1",["Array",16,"BitIn"]],
          ["O",["Array",16,"Bit"]],
          ["clk",["Named","coreir.clkIn"]],
          ["reset",["Named","coreir.arstIn"]],
          ["config_addr",["Array",1,"BitIn"]],
          ["config_data",["Array",2,"BitIn"]],
          ["config_en","BitIn"],
          ["read_data",["Array",2,"Bit"]]
        ]],
        "instances":{
          "const_0_1":{
            "genref":"coreir.const",
            "genargs":{"width":["Int",1]},
            "modargs":{"value":[["BitVector",1],"1'h0"]}
          },
          "inst0$inst0":{
            "genref":"coreir.reg_arst",
            "genargs":{"width":["Int",2]},
            "modargs":{"arst_posedge":["Bool",true], "clk_posedge":["Bool",true], "init":[["BitVector",2],"2'h0"]}
          },
          "inst0$inst1$inst0$_join":{
            "genref":"coreir.mux",
            "genargs":{"width":["Int",2]}
          },
          "inst1":{
            "genref":"coreir.eq",
            "genargs":{"width":["Int",1]}
          },
          "inst2":{
            "modref":"corebit.and"
          },
          "inst3$inst0":{
            "genref":"coreir.add",
            "genargs":{"width":["Int",16]}
          },
          "inst3$inst1$bit_const_1_None":{
            "modref":"corebit.const",
            "modargs":{"value":["Bool",true]}
          },
          "inst3$inst1$inst0":{
            "genref":"coreir.not",
            "genargs":{"width":["Int",16]}
          },
          "inst3$inst1$inst1$bit_const_0_None":{
            "modref":"corebit.const",
            "modargs":{"value":["Bool",false]}
          },
          "inst3$inst1$inst1$inst0":{
            "genref":"coreir.add",
            "genargs":{"width":["Int",16]}
          },
          "inst3$inst1$inst1$inst1":{
            "genref":"coreir.add",
            "genargs":{"width":["Int",16]}
          },
          "inst3$inst2":{
            "genref":"coreir.and",
            "genargs":{"width":["Int",16]}
          },
          "inst3$inst3":{
            "genref":"coreir.or",
            "genargs":{"width":["Int",16]}
          },
          "inst3$inst4$inst0$_join":{
            "genref":"coreir.mux",
            "genargs":{"width":["Int",16]}
          },
          "inst3$inst4$inst0$muxN_0$_join":{
            "genref":"coreir.mux",
            "genargs":{"width":["Int",16]}
          },
          "inst3$inst4$inst0$muxN_1$_join":{
            "genref":"coreir.mux",
            "genargs":{"width":["Int",16]}
          },
          "inst3$inst4$inst0$sel_slice0":{
            "genref":"coreir.slice",
            "genargs":{"hi":["Int",1], "lo":["Int",0], "width":["Int",2]}
          },
          "inst3$inst4$inst0$sel_slice1":{
            "genref":"coreir.slice",
            "genargs":{"hi":["Int",1], "lo":["Int",0], "width":["Int",2]}
          }
        },
        "connections":[
          ["inst1.in0","const_0_1.out"],
          ["self.reset","inst0$inst0.arst"],
          ["self.clk","inst0$inst0.clk"],
          ["inst0$inst1$inst0$_join.out","inst0$inst0.in"],
          ["inst0$inst1$inst0$_join.in0","inst0$inst0.out"],
          ["inst3$inst4$inst0$sel_slice0.in","inst0$inst0.out"],
          ["inst3$inst4$inst0$sel_slice1.in","inst0$inst0.out"],
          ["self.read_data","inst0$inst0.out"],
          ["inst3$inst4$inst0$_join.sel","inst0$inst0.out.1"],
          ["self.config_data","inst0$inst1$inst0$_join.in1"],
          ["inst2.out","inst0$inst1$inst0$_join.sel"],
          ["self.config_addr","inst1.in1"],
          ["inst2.in1","inst1.out"],
          ["self.config_en","inst2.in0"],
          ["self.I0","inst3$inst0.in0"],
          ["self.I1","inst3$inst0.in1"],
          ["inst3$inst4$inst0$muxN_0$_join.in0","inst3$inst0.out"],
          ["inst3$inst1$inst1$inst1.in0.0","inst3$inst1$bit_const_1_None.out"],
          ["self.I1","inst3$inst1$inst0.in"],
          ["inst3$inst1$inst1$inst0.in1","inst3$inst1$inst0.out"],
          ["inst3$inst1$inst1$inst1.in0.10","inst3$inst1$inst1$bit_const_0_None.out"],
          ["inst3$inst1$inst1$inst1.in0.11","inst3$inst1$inst1$bit_const_0_None.out"],
          ["inst3$inst1$inst1$inst1.in0.12","inst3$inst1$inst1$bit_const_0_None.out"],
          ["inst3$inst1$inst1$inst1.in0.13","inst3$inst1$inst1$bit_const_0_None.out"],
          ["inst3$inst1$inst1$inst1.in0.14","inst3$inst1$inst1$bit_const_0_None.out"],
          ["inst3$inst1$inst1$inst1.in0.15","inst3$inst1$inst1$bit_const_0_None.out"],
          ["inst3$inst1$inst1$inst1.in0.1","inst3$inst1$inst1$bit_const_0_None.out"],
          ["inst3$inst1$inst1$inst1.in0.2","inst3$inst1$inst1$bit_const_0_None.out"],
          ["inst3$inst1$inst1$inst1.in0.3","inst3$inst1$inst1$bit_const_0_None.out"],
          ["inst3$inst1$inst1$inst1.in0.4","inst3$inst1$inst1$bit_const_0_None.out"],
          ["inst3$inst1$inst1$inst1.in0.5","inst3$inst1$inst1$bit_const_0_None.out"],
          ["inst3$inst1$inst1$inst1.in0.6","inst3$inst1$inst1$bit_const_0_None.out"],
          ["inst3$inst1$inst1$inst1.in0.7","inst3$inst1$inst1$bit_const_0_None.out"],
          ["inst3$inst1$inst1$inst1.in0.8","inst3$inst1$inst1$bit_const_0_None.out"],
          ["inst3$inst1$inst1$inst1.in0.9","inst3$inst1$inst1$bit_const_0_None.out"],
          ["inst3$inst1$inst1$inst1.out","inst3$inst1$inst1$inst0.in0"],
          ["inst3$inst4$inst0$muxN_0$_join.in1","inst3$inst1$inst1$inst0.out"],
          ["self.I0","inst3$inst1$inst1$inst1.in1"],
          ["self.I0","inst3$inst2.in0"],
          ["self.I1","inst3$inst2.in1"],
          ["inst3$inst4$inst0$muxN_1$_join.in0","inst3$inst2.out"],
          ["self.I0","inst3$inst3.in0"],
          ["self.I1","inst3$inst3.in1"],
          ["inst3$inst4$inst0$muxN_1$_join.in1","inst3$inst3.out"],
          ["inst3$inst4$inst0$muxN_0$_join.out","inst3$inst4$inst0$_join.in0"],
          ["inst3$inst4$inst0$muxN_1$_join.out","inst3$inst4$inst0$_join.in1"],
          ["self.O","inst3$inst4$inst0$_join.out"],
          ["inst3$inst4$inst0$sel_slice0.out.0","inst3$inst4$inst0$muxN_0$_join.sel"],
          ["inst3$inst4$inst0$sel_slice1.out.0","inst3$inst4$inst0$muxN_1$_join.sel"]
        ]
      }
    }
  }
}
```
It's hard to see, but you can verify it in vim by placing your cursor on the `{` before `"top"` and using the `%` movement to go to the matching brace. There is none. Then if you go to the `{` after `"namespaces":` and use `%`, it jumps to the last `}` in the file.

rdaly525, 2018-10-11:
Well that is super strange. Ill take a look

rdaly525, 2018-10-11:
@leonardt How blocking is this?

leonardt, 2018-10-11:
I managed to work around it by just not running the `cullgraph` pass. I was running this to cleanup the output file after flattening (so I could easily give the file to Cristian for debugging CoSA). The culling of the graph is not necessary for the CoSA test in garnet (it works on the unculled graph), the output file just contains more noise, so it's not blocking at the moment, but may make debugging a bit harder in the future if we can't cull the graph.

rdaly525, 2018-10-30:
Fixed in #637

closed 2018-10-31 by rdaly525