rdaly525/coreir #647: Demosaic fails coreir simulator

jeffsetter, 2018-12-13:
I have been creating hardware for demosaic, but I am facing an issue when trying to simulate it during the circuit reset. Below is the error message. Any ideas on what could be happening?

Attached is the design:
[demosaic_design_top.txt](https://github.com/rdaly525/coreir/files/2677853/demosaic_design_top.txt)

starting reset
ERROR: add_882_884_885.in1 cannot be found

../../../../coreir/lib/libcoreir.so(_ZN6CoreIR14SimulatorState9getBitVecEPNS_6SelectE+0x5d)[0x2b7cc28933af]
../../../../coreir/lib/libcoreir.so(_ZN6CoreIR14SimulatorState17updateBitVecBinopEiPFN4bsim21quad_value_bit_vectorERKS2_S4_E+0x14d)[0x2b7cc28948ab]
../../../../coreir/lib/libcoreir.so(_ZN6CoreIR14SimulatorState13updateAddNodeEi+0x2f)[0x2b7cc2894ec1]
../../../../coreir/lib/libcoreir.so(_ZN6CoreIR14SimulatorState16updateNodeValuesEi+0x3bf)[0x2b7cc28970ff]
../../../../coreir/lib/libcoreir.so(_ZN6CoreIR14SimulatorState16exeCombinationalEv+0x259)[0x2b7cc28999b9]
../../../../coreir/lib/libcoreir.so(_ZN6CoreIR14SimulatorState12resetCircuitEv+0x18)[0x2b7cc28995b6]
./run[0x405cea]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf5)[0x2b7cc34cef45]
./run[0x4054d9]
make: *** [out.png] Error 1

leonardt, 2018-12-14:
If it's possible, could you include the test bench driving the simulator (that causes the error above)? This would likely speed up resolution by providing a way to reproduce the issue locally.

Even better might be a simple test bench to isolate the issue (just the necessary sequence of actions to run into the reset issue).

dillonhuff, 2018-12-14:
My guess is that the problem is the simulator mishandling multidimensional arrays. Are you running the flattentypes pass before executing?

jeffsetter, 2018-12-19:
Yes, I'm doing flattentypes before executing. This is the code before failure:

```
CoreIRLoadLibrary_commonlib(c);
  if (!loadFromFile(c,"./design_top.json")) {
    std::cout << "Could not Load from json!!" << std::endl;
    c->die();
  }

  c->runPasses({"rungenerators", "flattentypes", "flatten", "wireclocks-coreir"});

  std::cout << "finished passes" << std::endl;
  Module* m = g->getModule("DesignTop");
  assert(m != nullptr);
  SimulatorState state(m);

  state.setValue("self.in_arg_3_0_0", BitVector(16));                                                                                                      
  state.setValue("self.reset", BitVector(1));
  state.setClock("self.clk", 0, 1);
  std::cout << "starting reset" << std::endl;
  state.resetCircuit();
```

dillonhuff, 2018-12-19:
@jeffsetter I just copy-pasted your code in to a unit test and ran it in coreir. I do not get a segfault when running reset on demosaic. What version of coreir are you on?

jeffsetter, 2018-12-20:
I just updated to master with no change in outcome.

dillonhuff, 2019-01-08:
Here is my preprocessed version of demosaic. It does not fail in simulation:

[demosaic_design_top_preprocessed.txt](https://github.com/rdaly525/coreir/files/2738671/demosaic_design_top_preprocessed.txt)

jeffsetter, 2019-01-09:
It appears there is an issue with my version of memories.hpp. Closing

closed 2019-01-09 by jeffsetter