rdaly525/coreir #664: Testbench fails on fresh checkout of dev?

dillonhuff, 2019-01-30:
@rdaly525 I fetched the upstream of coreir, checked out dev and tried to build the testbench with a new floating point test, but when I did ```make test``` I got a testbench failure. The output is below:

```bash
bash-3.2$ make test -j
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C src dylib
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C passes
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C analysis
make[3]: Nothing to be done for `all'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C transform
make[3]: Nothing to be done for `all'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C ir dylib
make[2]: Nothing to be done for `dylib'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C simulator dylib
make[2]: Nothing to be done for `dylib'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C coreir-c dylib
make[2]: Nothing to be done for `dylib'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C libs dylib
make[2]: Nothing to be done for `dylib'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C tests
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C unit
make[2]: Nothing to be done for `all'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C unit-c
make[2]: Nothing to be done for `all'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C ext 
make[2]: Nothing to be done for `all'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C float
make[2]: Nothing to be done for `all'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C simulator
Makefile:23: warning: overriding commands for target `build/simtests'
Makefile:16: warning: ignoring old commands for target `build/simtests'
make[2]: Nothing to be done for `all'.
cd magma/inline && ls && ./run_tests.sh
conftest.py			run_tests.sh			test_const.v			test_simple_top.json		test_two_ops.json
gold				test_const.json			test_inline_operators.py	test_simple_top.v		test_two_ops.v
coreir.cpp:186 Running Runningvpasses
Numargs=1
In Run Generators
Done running generators
Numargs=1
Numargs=1
Numargs=1
Numargs=1
Numargs=2
Numargs=1
Numargs=2
Creating verilog for coreir.add
Creating verilog for global.Top
coreir.cpp:192 Running vpasses
coreir.cpp:228 Modified?: No
test_simple_top.v gold/test_simple_top.v differ: char 1, line 1
coreir.cpp:186 Running Runningvpasses
Numargs=1
In Run Generators
Done running generators
Numargs=1
Numargs=1
Numargs=1
Numargs=1
Numargs=2
Numargs=1
Numargs=2
Creating verilog for corebit.const
Creating verilog for coreir.add
Creating verilog for coreir.not
Creating verilog for global.Add8_cin
Creating verilog for global.Sub8
Creating verilog for global.test_two_ops
coreir.cpp:192 Running vpasses
coreir.cpp:228 Modified?: No
test_two_ops.v gold/test_two_ops.v differ: char 1, line 1
coreir.cpp:186 Running Runningvpasses
Numargs=1
In Run Generators
Done running generators
Numargs=1
Numargs=1
Numargs=1
Numargs=1
Numargs=2
Numargs=1
Numargs=2
Creating verilog for coreir.add
Creating verilog for coreir.const
Creating verilog for coreir.mul
Creating verilog for global.test_const
coreir.cpp:192 Running vpasses
coreir.cpp:228 Modified?: No
test_const.v gold/test_const.v differ: char 1, line 1
make[1]: *** [all] Error 1
make: *** [test] Error 2

```

Any idea whats wrong?

rdaly525, 2019-01-30:
Probably an old gold file. I will look into it.

rdaly525, 2019-02-12:
I cannot reproduce.

```
>cd coreir
>git checkout dev
>get pull
>make clean; make uninstall;
>make -j
>make test

```
This seems to pass for me.