rdaly525/coreir #665: Non determinism in verilog backend

leonardt, 2019-01-31:
```bash
cd coreir
git checkout determinism
make clean
make install -j
pip install delegator.py pytest
while pytest tests/binary/test_concat.py; do :; done  # should eventually fail
```

Contents of test_concat.py
```python
import os
import delegator

def test_concat_example():
    res = delegator.run('coreir -i examples/concat.json    '
                        '       -o tests/binary/build/out.v')
    assert not res.return_code, res.out + res.err
    res = delegator.run('diff tests/binary/build/out.v  '
                        '     tests/binary/gold/concat.v')
    assert not res.return_code, res.out + res.err
```

Example log of test failure:
```
~/repos/coreir determinism*
❯ while pytest tests/binary/test_concat.py; do :; done
/Users/lenny/miniconda3/lib/python3.7/site-packages/pep8.py:110: FutureWarning: Possible nested set at position 1
  EXTRANEOUS_WHITESPACE_REGEX = re.compile(r'[[({] | []}),;:]')
============================================== test session starts ===============================================
platform darwin -- Python 3.7.0, pytest-3.9.3, py-1.7.0, pluggy-0.8.0
rootdir: /Users/lenny/repos/coreir, inifile:
plugins: profiling-1.4.0, pep8-1.0.6, cov-2.6.0, codestyle-1.4.0
collected 1 item

tests/binary/test_concat.py .                                                                              [100%]

============================================ 1 passed in 0.05 seconds ============================================
/Users/lenny/miniconda3/lib/python3.7/site-packages/pep8.py:110: FutureWarning: Possible nested set at position 1
  EXTRANEOUS_WHITESPACE_REGEX = re.compile(r'[[({] | []}),;:]')
============================================== test session starts ===============================================
platform darwin -- Python 3.7.0, pytest-3.9.3, py-1.7.0, pluggy-0.8.0
rootdir: /Users/lenny/repos/coreir, inifile:
plugins: profiling-1.4.0, pep8-1.0.6, cov-2.6.0, codestyle-1.4.0
collected 1 item

tests/binary/test_concat.py .                                                                              [100%]

============================================ 1 passed in 0.05 seconds ============================================
/Users/lenny/miniconda3/lib/python3.7/site-packages/pep8.py:110: FutureWarning: Possible nested set at position 1
  EXTRANEOUS_WHITESPACE_REGEX = re.compile(r'[[({] | []}),;:]')
============================================== test session starts ===============================================
platform darwin -- Python 3.7.0, pytest-3.9.3, py-1.7.0, pluggy-0.8.0
rootdir: /Users/lenny/repos/coreir, inifile:
plugins: profiling-1.4.0, pep8-1.0.6, cov-2.6.0, codestyle-1.4.0
collected 1 item

tests/binary/test_concat.py F                                                                              [100%]

==================================================== FAILURES ====================================================
______________________________________________ test_concat_example _______________________________________________

    def test_concat_example():
        res = delegator.run('coreir -i examples/concat.json    '
                            '       -o tests/binary/build/out.v')
        assert not res.return_code, res.out + res.err
        res = delegator.run('diff tests/binary/build/out.v  '
                            '     tests/binary/gold/concat.v')
>       assert not res.return_code, res.out + res.err
E       AssertionError: 1,8d0
E         < module coreir_slice #(parameter hi=1, parameter lo=0, parameter width=1) (
E         <   input [width-1:0] in,
E         <   output [hi-lo-1:0] out
E         < );
E         <   assign out = in[hi-1:lo];
E         <
E         < endmodule  // coreir_slice
E         <
E         17a10,17
E         > module coreir_slice #(parameter hi=1, parameter lo=0, parameter width=1) (
E         >   input [width-1:0] in,
E         >   output [hi-lo-1:0] out
E         > );
E         >   assign out = in[hi-1:lo];
E         >
E         > endmodule  // coreir_slice
E         >
E
E       assert not 1
E        +  where 1 = <Command 'diff tests/binary/build/out.v       tests/binary/gold/concat.v'>.return_code

tests/binary/test_concat.py:11: AssertionError
============================================ 1 failed in 0.08 seconds ============================================
```

rdaly525, 2019-02-03:
@leonardt Just commited a fix into the determinism branch

rdaly525, 2019-02-03:
Sorry, this was not a good fix. Need to fix #669 before this one will work.

rdaly525, 2019-02-03:
Fixed by #668