rdaly525/coreir #676: Missing Generator Symbol: commonlib.muxn

scottcevans8, 2019-02-07:
Was having problems with trying to get the magma_tutorial working as it silently dies during compile and run. I made a smaller case: test.py

from magma_tutorial.adder import DefineAdd_
import fault
width = 4
Add = DefineAdd_(width)
tester = fault.Tester(Add)
tester.compile(target="verilator", directory="tests/build")

when I run it, it fails at
magma:WARNING:`m.compile` called with `output == verilog` and `m.mantle_target == "coreir"` and mantle has been imported, When generating verilog from circuits from the "coreir" mantle target, you should set `output="coreir-verilog"`. Doing this automatically.
Numargs=1
ERROR: In file: tests/build/Add4.json
Missing Generator Symbol: commonlib.muxn

Any pointers as to what I may be doing wrong would be appreciated.
Thanks

rdaly525, 2019-02-07:
I can look into this tomorrow. @leonardt, do you know if the magma coreir backend loads commonlib at any point?

scottcevans8, 2019-02-07:
Thank you. I'm not in a hurry, just wanted to take a look at the generated
files to understand a bit more about the tools.

leonardt, 2019-02-07:
Hmm, weird. The magma coreir backend will load commonlib if a magma circuit is compiled that references that coreir library. AFAIK, only commonlib.muxn in mantle uses this feature, and the DefineAdd_ example from the tutorial doesn't have that.

This may be related to the uniquification logic which was eager and compiling all circuits (so the muxn was getting compiled), the latest master fixes this issue.

leonardt, 2019-02-07:
Confirmed that the above test case is working for me on the latest magma master/release (v0.1.14).
@scottcevans8 when you get a chance, can you try upgrading your magma installation and running the test again?

scottcevans8, 2019-02-07:
I just installed 0.1.14 (was at 0.1.9 which I installed 2 days ago)

Have the same issue though although all of the magma tests are running now
for me. Before, several were failing.

I see in the Add4.json file that gets generated for my test case that there
is reference to the commonlib.muxn. I've attached the file to see if it
helps.

leonardt, 2019-02-07:
Seems like the attached file got dropped (perhaps GitHub doesn't support emailing attachments for issues). You can send it to lenny@stanford.edu if that's easier, or upload via the GitHub web interface.

Can you also provide what version of coreir you're using? If you built from source, just the commit sha of your repository checkout works.