rdaly525/coreir #678: Collapsing inline operators seems to not work

rsetaluri, 2019-02-07:
See the following magma:
```import magma as m
import mantle

def mux_ifc(height, width):
    ports = []
    for i in range(height):
        ports += [f"I{i}", m.In(m.Bits(width))]
    sel_bits = m.bitutils.clog2(height)
    ports += ["S", m.In(m.Bits(sel_bits))]
    ports += ["O", m.Out(m.Bits(width))]
    return ports

def gen(height, width):
    cell = m.DefineCircuit("aoimux_cell", *mux_ifc(2, 1))
    m.wire(cell.O, ((cell.I0 & ~cell.S) | (cell.I1 & cell.S)))
    m.EndDefine()
    return cell

m.compile("cell", gen(2, 1), output="coreir-verilog", inline=True)
```

We would expect the verilog to collapse the operators into `((I0 & ~S) | (I1 & S))` but it does not.

leonardt, 2019-02-08:
Yes the tests seem to be failing silently on travis (see https://github.com/rdaly525/coreir/issues/677). I'll investigate, there was likely a regression somewhere, possibly related to the debug verilator changes.

leonardt, 2020-08-25:
Just tested this on the latest master and it produces:

```
module aoimux_cell (
    input [0:0] I0,
    input [0:0] I1,
    input [0:0] S,
    output [0:0] O
);
assign O = (I0 & (~ S)) | (I1 & S);
endmodule
```

closed 2020-08-25 by leonardt