rdaly525/coreir #679: Simulation Semantics confusion

rdaly525, 2019-02-12:
@dillonhuff I am a little confused on the simulator semantics for the run() function and the runHalfCycle() (in interpret.cpp)

runHalfCycle() calls execute() first, then changes the value of the clock. This means that the evaluation of the new clock does not actually occur till the next time execute is called. Is this intentional, or is this expected behavior?

rdaly525, 2019-02-12:
My usual verilator test bench looks something like this to avoid the issue of changing the value of any input at the same time as changing the clock edge
```

for (...) {
   //Posedge just occured
  set_all_inputs();
  eval();
  clk = 0;
  eval();
  //Negedge just occured
  read_all_outputs();
  clk = 1;
  eval();
}

```

dillonhuff, 2019-02-12:
```
runHalfCycle() calls execute() first, then changes the value of the clock. This means that the evaluation of the new clock does not actually occur till the next time execute is called. Is this intentional, or is this expected behavior?
```

That is the intended behavior, though I don't remember why it was added.

dillonhuff, 2019-02-13:
@rdaly525 can I close this issue?

closed 2019-02-18 by dillonhuff