rdaly525/coreir #694: Add CoreIR Pass That Counts Primtive Usage

David-Durst, 2019-02-19:
Please add a pass that counts all the primitives that a design uses. For example, I would like to see all the 8 bit adders, 16 bit adders, n bit muxes, etc. 

Lower priority part of request: if possible, could it be possible to run this hierarchically? For example, if the design hasn't been flattened, could I get the amount of primitives each instance requires?

rdaly525, 2019-02-19:
To reiterate what we talked about for the final format: For every level of module hierarch, for every coreir primitive, print out the number of instances of that primitive, and the total number of instances of that primitive for all the children.

David-Durst, 2019-02-19:
Yes, thank you!

rdaly525, 2019-02-25:
Done in #696

closed 2019-02-25 by rdaly525