rdaly525/coreir #712: Branch cleanup

rdaly525, 2019-03-14:
Profiling -Jeff
cansel -Lenny
mem-hack -Lenny

master_Freeze_8_21_18 -Ross
mnr -Ross
object -Ross
Simcheck -Ross
Devtest -Ross
virtualtg -Ross
448b-freeze -Ross
typelang -Ross

@leonardt, @jeffsetter, what is the state of your coreir branches here? Can you delete/do a PR or let me know if you still need the branch?

leonardt, 2019-03-14:
Deleted my branch.  IIRC, 448b-freeze was the version of coreir used for our visualization project? That can probably be deleted, you could instead use a git-tag for the commit if we need to reference it in the future.

jeffsetter, 2019-03-14:
Deleted profiling, since its implementation was later implemented for #696