rdaly525/coreir #727: Floating point support: CoreIR and simulator

jeffsetter, 2019-04-25:
Add support for all floating operators:
add, mul, sub
le, lt, ge, gt, neq, neq, neg, sqr, flr, ceil, abs, min, max
div, rem
log, exp, pow, sqrt
sin, cos, tan, asin, acos, atan2

These are listed in order from most important to least important.

dillonhuff, 2019-04-25:
Cool. I assume bfloat and 32 bit are sufficient for now?

jeffsetter, 2019-04-25:
Yes. BFloat and Float32 would be sufficient.

jeffsetter, 2019-05-08:
Btw, if you're interested in looking at the code that Halide uses to implement bfloat16, these two files that Andrew wrote are useful:
https://github.com/StanfordAHA/Halide-to-Hardware/blob/master/src/Float16.cpp
https://github.com/StanfordAHA/Halide-to-Hardware/blob/master/src/EmulateFloat16Math.cpp