rdaly525/coreir #730: Cannot connect Bit to BitInOut

leonardt, 2019-04-25:
When trying to connect a Bit to a BitInOut I get the following error:
```
ERROR: Top: Cannot wire together
  self.pad : Bit
  PRWDWUWSWCDGH_V_inst0.PAD : BitInOut
```

Is this expected behavior? Should I be using `wrap` to convert from Bit to BitInOut?

leonardt, 2019-04-25:
It seems like `wrap` is specifically for named types (since it uses `getRaw`)

rdaly525, 2019-04-25:
You have to use either the tribuf primitive or the ibuf primitive depending on what you are trying to do. Tribuf is for driving an Inout and ibuf is reading from an ibuf. Sounds like you want the latter.

leonardt, 2019-04-25:
Here's the specific use case:

We have a verilog file `pad.v`
```
module PRWDWUWSWCDGH_V (C, DS0, DS1, DS2, I, IE, OEN, PAD, PU, PD, ST, SL, RTE);

  input  DS0, DS1, DS2, I, IE, OEN, PU, PD, ST, SL, RTE;
  output C;
  inout  PAD;

  assign C = PAD;

  assign pad = ((IE == 1'b0) & (OEN ==1'b0)) ? I : 1'bz;

endmodule
```

we're trying to do something in magma like

```
    Pad = m.DeclareFromVerilogFile(os.path.join(file_path, "pad.v"))[0]

    class Top(m.Circuit):
        IO = ["pad", m.In(m.Bit)]
        @classmethod
        def definition(io):
            pad = Pad()
            m.wire(io.pad, pad.PAD)
            for port in ["DS0", "DS1", "DS2", "I", "IE", "OEN", "PU", "PD", "ST", "SL", "RTE"]:
                m.wire(0, getattr(pad, port))
```

In this case, should the magma be user be inserting tribufs?

rdaly525, 2019-04-25:
You would have to use an “ibuf” for “assign c = PAD”, and a tribuf for “assign PAD = ...”

Both of these will be structural instances. When we have verilog inlining working, it will resolve to what you have written.

rdaly525, 2019-04-25:
Oh sorry I misread the question. This holds if you wanted to describe the verilog file using coreir.

If you want to “read” from pad.PAD, you have to use an ibuf in coreir

leonardt, 2019-04-25:
We're not trying to recreate the verilog in magma, the magma code is trying to instance the above verilog module (wrapped) which has an inout port in the interface. Magma allows connecting an output to an inout, but then the coreir backend fails when trying to compile that. Either magma could insert the `ibuf` or we could change magma to match coreir by requiring an explicit cast with `ibuf` or `tribuf`.  When I look at `tribuf` though, it seems like I need to provide an `en` signal to use it, but in this case, the enable logic is handled by the wrapped verilog module, so there's not really a reasonable signal for us to use.

leonardt, 2019-04-25:
We have another simple example where we're trying to connect an input to the wrapped verilog module, so we also need to "write" to the pad. That means we need to use `tribuf`, but what do we use for the `en` signal? I guess 1 could work?

```
    Pad = m.DeclareFromVerilogFile(os.path.join(file_path, "pad.v"))[0]

    class Top(m.Circuit):
        IO = ["pad", m.Out(m.Bit)]
        @classmethod
        def definition(io):
            pad = Pad()
            m.wire(io.pad, pad.PAD)
            for port in ["DS0", "DS1", "DS2", "I", "IE", "OEN", "PU", "PD", "ST", "SL", "RTE"]:
                m.wire(0, getattr(pad, port))

    m.compile("build/test_pad", Top, output="coreir-verilog")
```

leonardt, 2019-04-25:
(I think I had the examples flipped in terms of read/write)

leonardt, 2019-04-25:
Alex confirmed that the pad already contains the tri-state logic (this is actually a stub for IP he's trying to wrap), so it seems like we need a solution for inouts that handles the case when the tri-state logic is inside a black box wrapped verilog module

rdaly525, 2019-04-25:
I don’t understand the context where you need to write to an IO without there being an enable signal. Usually an IO (or any inout) is used because there is a need for multiple drivers for that signal. Having multiple drivers requires there to be an enable signal for each driver of the Inout. If you are trying to test a case where it is guaranteed that there is only one driver of the IO and you have to use a predefined verilog module which had an inout declaration, then yeah, you should stick a constant 1 on the enable. But in general if there is only one driver, then there is no need to use an inout declaration.

rdaly525, 2019-04-25:
Usually for an IO pad that can act as either an input or an output, the enable signal for the tribuf comes from some configuration bit indicating whether you are treating the IO pad as an input or an output.

leonardt, 2019-04-25:
I think the use case is: Given an existing verilog module implementing a pad that has an `inout` in the interface and the enable logic is handled internally, how do we use this with coreir? It seems like your answer is to stick in a `tribuf` with a 1 on the enable? We can do that, but I'm not sure if that's the most elegant solution. Does this mean that the generated verilog will include this tribuf logic with a 1?

rdaly525, 2019-04-25:
Well it depends on how you want to interact with the inout signal. If you want to be able to drive that signal, you need to use a tribuf (with an enable). If you just want to read that signal (translate it into a BitOut), then coreir requires you to use an “ibuf” instance (which does not require an enable). Am I missing something?

leonardt, 2019-04-25:
How will this be compiled to verilog? Specifically, will the tribuf with enable 1 be lowered to verilog that inserts the enable logic? Is there a plan to remove that in this case where the enable is a constant 1? 

Are there any other alternative design patterns here that avoids introducing noise into the generated verilog or complexity in the verilog backend to remove this noise?

It seems like the reason to introduce `ibuf` and `tribuf` to handle connecting `inout`s is to enforce static safety checks when working with inouts. In this case, we don't need those checks because the logic related to inouts is hidden inside a black box module. One alternative design would be to introduce the notion of a `cast` or something that allows us to wire up inputs/outputs to an inout when we know it's safe to do.  `ibuf` is essentially this for outputs. The main issue here is that we're required to use `tribuf` with an enable, but it's not providing us any safety guarantees, so it's introducing extra complexity in the frontend (the user or magma has to insert these) without offering any advantages. 

For now we can move forward with wiring a 1 to the enable of the tribuf,  but I think this is a design issue worth considering unless we think this use case is "niche" and won't be a problem for most users.

alexcarsello, 2019-04-25:
The issue is that I have a pad cell which has tri-state logic inside it (and this has an inout pad port). I should not have to instantiate a tri-state buffer on the pad, which already contains a tri-state buffer. TheI should just be able to connect the pad pin the top-level pin of my chip. Even if I were using the pad bidirectionally, no circuit designer is placing an extra tri-state buffer on the pad cell which already contains the necessary tri-state logic.

Side note: there were also issues when I made the top-level pin an inout (tried to connect one inout to another).

leonardt, 2019-04-25:
@alexcarsello the issue connecting inouts to inouts is likely a magma issue (that may be fixed in the inout branch), let me take a look at that

rdaly525, 2019-04-25:
Perhaps the confusion is that I did not mention in coreir, you can connect an inout to an inout? There is no need to instantiate either a tribuf or an ibuf when just passing the inout wire through levels of instantiation.

You can think of ibuf as a kind of "cast" that allows for some form of static type checking. But the tribuf is a fundamental thing that semantically requires there to be an enable signal. There is really no use case where you would like to drive a signal onto an inout wire that does not require an enable signal. This is fundamentally because an inout signal implies that you can have multiple drivers for that inout signal. Having multiple drivers means you have to have an enable for each driver. 

For your use case, you are saying that the tribuf logic is happening already inside the black box verilog module. This means that you are probably not trying to drive the inout signal with any other signal, so there is no reason to instantiate another tribuf.

leonardt, 2019-04-25:
Here is Alex's `Generator` code

```
class Pad(Generator):
    def __init__(self, is_input, is_vertical):
        super().__init__()
        self.is_input = is_input
        self.is_vertical = is_vertical
        if self.is_input:
            pad_type = magma.In(magma.Bit)
            fabric_type = magma.Out(magma.Bit)
        else:
            pad_type = magma.Out(magma.Bit)
            fabric_type = magma.In(magma.Bit)
        self.add_ports(
            pad=pad_type,
            fabric=fabric_type,
            rte=magma.In(magma.Bit)
        )

        pad_verilog_file = "pad_wrapper/pad_stub"
        if self.is_vertical:
            pad_verilog_file += "_v.v"
        else:
            pad_verilog_file += "_h.v"

        self.pad = FromVerilog(pad_verilog_file)
        # Wire up input/output/control signals to pad
        self.wire(self.ports.pad, self.pad.ports.PAD)
        self.wire(Const(magma.Bit(self.is_input)), self.pad.ports.IE)
        self.wire(Const(magma.Bit(self.is_input)), self.pad.ports.OEN)
        self.wire(self.ports.rte, self.pad.ports.RTE)
        if self.is_input:
            self.wire(self.ports.fabric, self.pad.ports.C)
            self.wire(Const(magma.Bit(0)), self.pad.ports.I)
        else:
            self.wire(self.ports.fabric, self.pad.ports.I)

        # Other random signals we need to hook up
        self.wire(Const(magma.Bit(0)), self.pad.ports.ST)
        self.wire(Const(magma.Bit(0)), self.pad.ports.SL)
        self.wire(Const(magma.Bit(0)), self.pad.ports.DS0)
        self.wire(Const(magma.Bit(0)), self.pad.ports.DS1)
        self.wire(Const(magma.Bit(0)), self.pad.ports.DS2)
        self.wire(Const(magma.Bit(0)), self.pad.ports.PU)
        self.wire(Const(magma.Bit(0)), self.pad.ports.PD)
```

from what I understand (@alexcarsello can clarify or expand), certain pads on the CGRA are parametrized with `is_input`. The generator generates a wrapper around the pad IP that changes the `inout` to and `input` or `output` (not shown in the generator stub, but I'm guessing alex will introduce other logic to configure the pad appropriately). So, we have a black box verilog module with an `inout` that we're trying to wire up to either an `input` or and `output`. From what I understand, if the pad is configured as an input, we need to insert a tribuf in order to wire up the `input` of the generated module to the `inout` of wrapped verilog module.

alexcarsello, 2019-04-26:
@rdaly525 I am trying to drive the inout signal on the pad itself with the top level input of the chip, since the pad is configured to be in input mode permanently. In this case, I do not need to put a tri-state buffer on this top-level input to drive the pad.

leonardt, 2019-04-26:
I think one possible design alternative we can consider is introducing a special construct for working with Verilog `inout`s. It seems like coreir's use of `inout` is designed in a fashion that doesn't map cleanly to the verilog `inout`.

From what I understand, coreir's `inout` is designed such that the user is never allowed to connect directly to the `inout` and instead must go through the `ibuf` and `tribuf` primitives. Verilog on the other hand, supports directly connecting inputs and outputs to an inout.  The main issue with this semantic difference is that Verilog modules are then designed with `inout` ports with this behavior in mind (example shown by the pad that we're trying to integrate). In coreir, this pad would expose an input and an output port (for the tribuf/ibuf) and the user would use one depending on which configuration they chose (and presumably wire an unused constant or term to the other).

I think for now we can workaround this issue by inserting the tribuf (we'll have to decide whether the user or magma handles this) and trying to remove the code in the verilog backend (we see the enable is a constant 1 so we can remove the tribuf logic).  I'm not sure if there's a good solution to handling this semantic mismatch besides special casing it the verilog backend.

One corollary issue is: how would you write the above pad (verilog stub) in coreir? Would you expose both the input and outputs to the user and they have to select the one they want to use and wire up the unused port appropriately? Or perhaps the long term goal would be to have the pad defined as an implementation of the `tribuf`/`ibuf` primitives somehow?

leonardt, 2019-04-26:
Now that I think about it, I think for the short-term furture, anyone manufacturing a design with IO (presumably most designs) will be running into this issue because the pad block will almost certainly be vendor IP defined as black box verilog, so I think this is an essential design issue worth considering.

Perhaps we should step back and consider this use case from the top-down (rather than trying to figure out the best way to implement it given coreir's inout semantics).  What's the story for integrating a black box pad with an `inout` in the interface but tribuf logic internally? Right now, it seems that the user (or compiler) must insert `tribuf`s in order to wire an input to the pad. Is there an alternative way to handle this use case?

rdaly525, 2019-04-26:
So just so I am clear:

-You want each Pad to only ever by either an "input" (data flows from outside the chip to into the chip)
or an "output" (data flows from inside the chip to outside the chip)
-You want the port connecting the outside of the chip to be either an BitIn or a BitOut (But *not* an InOut)
-You want to use a blackbox verilog module that utilizes an inout port to connect outside the chip.

The weird thing here is that your "pad" signal is not of type inout, but you are using blackbox verilog that is treating it as an inout. I think that you will find if you try to manually create the verilog version of the "is_input=True" Pad, it will not compile. 

I tried to write the simplest version of what you want to do in verilog. The PadWrapper input does not compile because you are not allowed to directly connect a verilog input to an inout

```

module Pad(
  input I,
  input en,
  output o,
  inout pad
);

  assign pad = en ? I : 1'bz;
  assign o = pad;
endmodule

module PadWrapperOutput(
  output pad,
  input I,
  input en
);

  Pad pinst(
    .I(I),
    .en(en),
    .o(),
    .pad(pad)
  );
endmodule

module PadWrapperInput(
  input pad,
  output o
);

  Pad pinst(
    .I(1'b0),
    .en(1'b0),
    .o(o),
    .pad(pad)
  );
endmodule
```

alexcarsello, 2019-04-26:
I'm not sure what you used to try to compile this, but I can confirm that this does compile because it was used on the last chip, and on multiple other top-level netlists I've seen.

rdaly525, 2019-04-26:
Verilator does not allows this to compile. Again, semantically it really does not make sense. The highest level pad type really should be "inout" instead of input

rdaly525, 2019-04-26:
A wire declared as input inherently means that there is always a value driven on that wire. This is incompatible with inout (unless you use a tri state buffer). So if you really want this use case, I would claim it is a very niche and strange use case, and that you should stick in a tristate buffer with en=1 here to indicate this is what you are meaning to do. Perhaps an optimization pass can see that you have two tristate buffers driving the inout where one has en=0 always and the other has en=1 always and it will therefore completely get rid of the inout port. But I suspect you are using some physical design IP for the pad. I think manually inserting the tri state buffer with en=1 is the proper thing to do in this weird case.

Either that or change the type of pad to be inout.

rdaly525, 2019-04-26:
Okay, I think I can summarize the issue to reconcile everything.

Inouts inherently imply that they are capable of being driven by multiple signals. The normal use case to handle this is to only drive an inout with an enabled tristate buffer. The use case that you have in mind is that you have a module instantiation with an inout port which you would like to drive with an input that does not require an enable. This is only semantically meaningful because you can guarantee that nothing else will be driving this inout wire (because you directly connect the other tristate buffer's enable to 1'b0). In general this is not the case especially with a black box module since the en on the other tristate buffer's en *could* be 1 sometimes.

Verilog's support for inout in general is sketchy.

1) This does not compile (for verilator, but possibly does for other simulators/synthesis tools) 

```
module Pad(
  input I,
  input en,
  output o,
  inout pad
);

  assign pad = en ? I : 1'bz;
  assign o = pad;
endmodule

module PadWrapperInput(
  input pad,
  output o
);

  Pad pinst(
    .I(1'b0),
    .en(1'b0),
    .o(o),
    .pad(pad)
  );
endmodule
```

2) But this does:

```
module Pad(
  input I,
  input en,
  output o,
  inout pad
);

  assign pad = en ? I : 1'bz;
  assign o = pad;
endmodule

module PadWrapperInput(
  input pad,
  output o
);
  wire _pad;
  assign _pad = pad;

  Pad pinst(
    .I(1'b0),
    .en(1'b0),
    .o(o),
    .pad(_pad)
  );
endmodule
```

3) Also the only way to "connect" an inout to another inout is implicitly by name of port.

```
module PadWrapperInput(
  inout pad,
  output o
);

  Pad pinst(
    .I(1'b0),
    .en(1'b0),
    .o(o),
    .pad(pad)
  );
endmodule
```

4) There is no way to do something like the following:
```
module PadWrapperInput(
  inout pad,
  output o
);
  
  wire _pad;
  //Bidirectional connection??
  assign _pad = pad; 
  assign pad = _pad;

  Pad pinst(
    .I(1'b0),
    .en(1'b0),
    .o(o),
    .pad(_pad)
  );
endmodule
```

It sounds like you want a feature request for CoreIR to generate code that looks like 1) or 2)
CoreIR does not inherently support that since it only allows you to connect BitInOut to other BitInOut. This would require fundamentally changing the static type checking of CoreIR.

I think the best workaround here is to use a tribuf with en=1. Semantically this makes the most sense. You have two singals driving the inout, but you can guarantee that the first signal's enable is 0, therefore the second signal's enable can be 1.

After running verilog inlining  + simple verilog mux elimination (tribuf 'mux' in this case) will be able to translate it into 2). 

@leonardt, @alexcarsello, does this make sense?

rdaly525, 2019-04-26:
In terms of whether magma wants to support directly connecting an inout to an input, I am indifferent. I lean towards this being a weird case and that the tristate buffer should be manually added, but I have my biases.

leonardt, 2019-04-26:
I don't think we necessarily have to change CoreIR's static semantics, an alternative would be to introduce a notion of a cast or "unsafe" like rust. Basically, there's a case (albeit niche) where we know that this is safe to do, but generally coreir believes this is an unsafe thing to do so it strictly prevents it. Unsafe/cast would allow us to make this connection knowing that it's okay.

However, this is a non-blocking issue and more of an internal design issue (mainly to keep the compiler internals in magma and coreir simpler and easier to maintain). I think optimizing it out in the verilog code generation via inlining and mux elimination should work.

closed 2019-04-26 by leonardt