rdaly525/coreir #734: Enforce code style guidelines

leonardt, 2019-04-26:
I think it would be good to start enforcing coding style guidelines so the codebase is consistent. @rsetaluri would vote for https://google.github.io/styleguide/cppguide.html#Formatting. A simpler, automated option is to use https://clang.llvm.org/docs/ClangFormat.html.

I think we should at least run a tool like `clang-format` on the entire code base to get it into a consistent state. The benefit of using an automated approach is that we can simply enforce style by running the formatter before accepting PRs (so contributors are free to develop using whatever style they prefer).

rdaly525, 2019-04-26:
Agreed.

rsetaluri, 2019-04-29:
Ok with either, think they're pretty similar tbh.

Google also has a lot of tools for their style: https://github.com/google/styleguide