rdaly525/coreir #746: Floating point reset fails

jeffsetter, 2019-05-14:
I have a design that is failing reset. It seems to occur during `state.resetCircuit()` 

So far, it seems that fp_pointwise passes with fp_add or fp_mul; conv_3_3 works with int operators; however, fp_conv_3_3 fails during this step.

The failure is due to some value being filled with 'x's in extendBFloat, but I haven't figured out why one of the inputs of the floating point operator would have 'x's. Do you have any ideas on what is happening?

Below are the test design and simulation call (run_coreir_on_interpreter).
[coreir_interpret.cpp.txt](https://github.com/rdaly525/coreir/files/3180046/coreir_interpret.cpp.txt)
[fp_conv_3_3_design_top.json.txt](https://github.com/rdaly525/coreir/files/3180054/fp_conv_3_3_design_top.json.txt)

dillonhuff, 2019-05-15:
@jeffsetter I have no idea why resetCircuit would fail because of a floating point op.

How did you figure out that the failure is caused by 'x's in extendBFloat? Could you send a trace of what happens in the simulator?

jeffsetter, 2019-05-15:
Below is a trace from gdb:
Note that I uncommented
https://github.com/rdaly525/coreir/blob/dev/src/simulator/interpret.cpp#L31
https://github.com/rdaly525/coreir/blob/dev/src/simulator/interpret.cpp#L58

Starting program: /nobackup/setter/h2h/float_tests/Halide-to-Hardware_float/apps/hardware_benchmarks/tests/fp_conv_3_3/bin/process run coreir input.png
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
/nobackup/setter/h2h/float_tests/coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/nobackup/setter/h2h/float_tests/coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
Found raddr
Found raddr
Starting topological sort
topo_order.size() = 183
numVertices(g)    = 183
generated simulated coreir design
reset clock self.clk
reset self.in_arg_0_0_0 with size 16
reset self.reset with size 1
image is using output valid
starting coreir simulation
bfloat = 0000000000000000
Extension result = 00000000000000000000000000000000
bfloat = 0100000001111001
Extension result = 01000000011110010000000000000000
bfloat = xxxxxxxxxxxxxxxx
Extension result = 0xxxxxxxxxxxxxxx0000000000000000
bfloat = 0000000000000000
Extension result = 00000000000000000000000000000000
process: ../../../../../coreir/include/coreir/ir/dynamic_bit_vector.h:123: unsigned char bsim::quad_value::binary_value() const: Assertion `(value == 1) || (value == 0)' failed.

Program received signal SIGABRT, Aborted.
0x00007ffff584c428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54  ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff584c428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff584e02a in __GI_abort () at abort.c:89
#2  0x00007ffff5844bd7 in __assert_fail_base (fmt=<optimized out>, assertion=assertion@entry=0x53c766 "(value == 1) || (value == 0)",
    file=file@entry=0x53c4c0 "../../../../../coreir/include/coreir/ir/dynamic_bit_vector.h", line=line@entry=123,
    function=function@entry=0x53cec0 <bsim::quad_value::binary_value() const::__PRETTY_FUNCTION__> "unsigned char bsim::quad_value::binary_value() const") at assert.c:92
#3  0x00007ffff5844c82 in __GI___assert_fail (assertion=0x53c766 "(value == 1) || (value == 0)", file=0x53c4c0 "../../../../../coreir/include/coreir/ir/dynamic_bit_vector.h", line=123,
    function=0x53cec0 <bsim::quad_value::binary_value() const::__PRETTY_FUNCTION__> "unsigned char bsim::quad_value::binary_value() const") at assert.c:101
#4  0x0000000000517b29 in bsim::quad_value::binary_value() const ()
#5  0x0000000000521512 in int bsim::quad_value_bit_vector::to_type<int>() const ()
#6  0x00007ffff69ee791 in CoreIR::SimulatorState::updateNodeValues(int)::{lambda(bsim::quad_value_bit_vector const&, bsim::quad_value_bit_vector const&)#22}::operator()(bsim::quad_value_bit_vector cons$
#7  0x00007ffff69ee936 in CoreIR::SimulatorState::updateNodeValues(int)::{lambda(bsim::quad_value_bit_vector const&, bsim::quad_value_bit_vector const&)#22}::_FUN(bsim::quad_value_bit_vector const&, bs$
#8  0x00007ffff69eb4a1 in CoreIR::SimulatorState::updateBitVecBinop(int, bsim::quad_value_bit_vector (*)(bsim::quad_value_bit_vector const&, bsim::quad_value_bit_vector const&)) ()
   from ../../../../../coreir/lib/libcoreirsim.so
#9  0x00007ffff69f0b43 in CoreIR::SimulatorState::updateNodeValues(int) () from ../../../../../coreir/lib/libcoreirsim.so
#10 0x00007ffff69f32f5 in CoreIR::SimulatorState::exeCombinational() () from ../../../../../coreir/lib/libcoreirsim.so
#11 0x00007ffff69f2e84 in CoreIR::SimulatorState::resetCircuit() () from ../../../../../coreir/lib/libcoreirsim.so
#12 0x0000000000519e22 in void run_coreir_on_interpreter<unsigned short>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, Halide::Runtime::Buffer<unsigned short, 4>, Hal$
#13 0x0000000000418867 in main::{lambda()#2}::operator()() const ()
#14 0x0000000000435fb6 in std::function<void ()>::operator()() const ()
#15 0x000000000042e86e in OneInOneOut_ProcessController<unsigned short>::make_run_def(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__$
#16 0x000000000042e0a5 in ProcessController<unsigned short>::process_command(int, char**) ()
#17 0x0000000000417f33 in main ()

dillonhuff, 2019-05-15:
Looking at this trace I think I understand the issue. The calls to float ops use casting of BitVector -> ```int``` in their implementation (so that they can then cast the ```int``` to a  ```float``` and use the CPUs floating point to emulate bfloats). This casting routine does not handle 'x's.

I will add a check for x values and return x if any float value contains an 'x'.

jeffsetter, 2019-05-15:
Do I want any 'x's though? During reset, I wasn't expecting any 'x's to be present in my design.

dillonhuff, 2019-05-15:
My understanding is that without x's on wires the moment before reset we will not be able to match Verilog simulation semantics, so having them is important.

What is your concern with having 'x' values on wires?

dillonhuff, 2019-05-15:
Aso I just pushed an update to #748 that should fix this casting issue.

jeffsetter, 2019-05-16:
Everything seems to be fixed based on https://github.com/rdaly525/coreir/pull/748

closed 2019-05-16 by jeffsetter