rdaly525/coreir #747: Floating point sub

jeffsetter, 2019-05-14:
It appears that floating point subtraction is not working.

Subtracting seems to be adding according to my tests, which can also be seen by using this test: https://github.com/rdaly525/coreir/blob/dev/tests/simulator/interpreter.cpp#L790

dillonhuff, 2019-05-15:
@jeffsetter on closer examination that subtraction test case was broken (I was not truncating the 32 bit float result to 16 bits in the testbench.)

I have added more unit tests for both 16 and 32 bit and the tests pass locally. Are you sure something is wrong in the subtract node?

jeffsetter, 2019-05-15:
My test case was doing essentially input(x,y) + -pi (I thought it was subtraction, when in reality it was adding a negative number). Below is the output:
```finished running CoreIR code
Ran fp_pointwise on coreir
input(0,0) = 0   -pi = 3
input(1,0) = 1   -pi = 4
input(2,0) = 2   -pi = 5
input(3,0) = 3   -pi = 6
input(4,0) = 4   -pi = 7
input(5,0) = 5   -pi = 8
input(6,0) = 6   -pi = 9
input(7,0) = 7   -pi = 10
input(8,0) = 8   -pi = 11
input(9,0) = 9   -pi = 12
```
 For some reason it was not looking at the sign bit properly. 
Looking at the implementation a little more closely, the error is fixed below:
```diff --git a/src/simulator/interpret.cpp b/src/simulator/interpret.cpp
index 9112f5a..8be7981 100644
--- a/src/simulator/interpret.cpp
+++ b/src/simulator/interpret.cpp
@@ -32,7 +32,7 @@ namespace CoreIR {
     assert(r.bitLength() == 16);

     BitVector sgn(1, 0);
-    sgn.set(1, r.get(15));
+    sgn.set(0, r.get(15));

     BitVector exp = slice(r, 7, 15);
     assert(exp.bitLength() == 8);
```
Implementing this fix, it is now passing. It appears the test cases were not hitting this case.

dillonhuff, 2019-05-15:
I just added that fix on my pull request. Thanks!

jeffsetter, 2019-05-15:
In addition, this could be caught in BitVector if it ensures that indices are not set out of bounds. 
https://github.com/rdaly525/coreir/blob/dev/include/coreir/ir/dynamic_bit_vector.h#L514

We could at a ```.at()``` or an assert that the index is less than the array size.

jeffsetter, 2019-05-15:
This should be fixed in https://github.com/rdaly525/coreir/pull/748

closed 2019-05-15 by jeffsetter