rdaly525/coreir #760: ROM support in coreir simulator

dillonhuff, 2019-06-12:
@Kuree I remember you saying something about the need for ROMs in the coreir simulator?

Do you have any examples of ROM use as a guide for unit testing?

Kuree, 2019-06-12:
Here: https://github.com/StanfordAHA/Halide-to-Hardware/tree/master/apps/hardware_benchmarks/tests/rom

dillonhuff, 2019-06-20:
@Kuree I just got an email from Mark saying that camera pipeline is not working in the simulator.

Is the lack of ROM support what is blocking camera pipeline? If not could you file a separate issue about it.

Kuree, 2019-06-20:
According to Jeff, rom is the only thing that's blocking the simulation.

dillonhuff, 2019-06-21:
@jeffsetter could you send me the simulator output when you run on camera pipeline? It would be helpful to see the actual failure.

dillonhuff, 2019-06-21:
@Kuree we now have simulator support for ROMs in #764 . I could not get the example you linked to to build in Halide-to-Hardware (for other reasons related to my local machine settings), so I just tried to guess about the json initialization format. If there are other issues then @jeffsetter please send me the camera pipe json file and I can look at the ROM there.

Kuree, 2019-06-21:
@dillonhuff I don't know how to run the simulator, but here are the files you need:
- camera pipeline json file: https://transfer.sh/BM7IS/design_top.json
- input file: https://transfer.sh/FND1E/input.raw
- gold file: https://transfer.sh/cDPd4/output_cpu.raw

You should be able to use `wget` to download the files directly from the link.

dillonhuff, 2019-06-21:
Thanks for sending those.

@rdaly525 I notice that in the generated json rom init values are in decimal. Was there any reason not to use bit vectors?

dillonhuff, 2019-06-21:
@rdaly525 @Kuree changed the PR to assume ROM defaults will be encoded in decimal

jeffsetter, 2019-06-21:
Not a big reason not to use bitvector. We could change from decimal to bitvector.

jeffsetter, 2019-06-21:
Starting program: /nobackup/setter/h2h/float_tests/Halide-to-Hardware_float/apps/hardware_benchmarks/apps/camera_pipeline/bin/process run coreir input.png
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
/nobackup/setter/h2h/float_tests/coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/nobackup/setter/h2h/float_tests/coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
Found raddr
Found raddr
Found raddr
Found raddr
Found raddr
Found raddr
Found raddr
Found raddr
Found raddr
Starting topological sort
topo_order.size() = 916
numVertices(g)    = 916
generated simulated coreir design
reset clock self.clk
reset self.in_arg_3_0_0 with size 16
reset self.in_en with size 1
reset self.reset with size 1
image is using output valid
starting coreir simulation
ERROR: rom__curvea0$mem.raddr cannot be found

../../../../../coreir/lib/libcoreirsim.so(_ZN6CoreIR14SimulatorState9getBitVecEPNS_6SelectE+0x72)[0x7ffff69d740a]
../../../../../coreir/lib/libcoreirsim.so(_ZN6CoreIR14SimulatorState18updateMemoryOutputEi+0xf8)[0x7ffff69dc8ce]
../../../../../coreir/lib/libcoreirsim.so(_ZN6CoreIR14SimulatorState16exeCombinationalEv+0x140)[0x7ffff69dea2c]
../../../../../coreir/lib/libcoreirsim.so(_ZN6CoreIR14SimulatorState12resetCircuitEv+0x18)[0x7ffff69de734]
/nobackup/setter/h2h/float_tests/Halide-to-Hardware_float/apps/hardware_benchmarks/apps/camera_pipeline/bin/process[0x5376db]
/nobackup/setter/h2h/float_tests/Halide-to-Hardware_float/apps/hardware_benchmarks/apps/camera_pipeline/bin/process[0x418b74]
/nobackup/setter/h2h/float_tests/Halide-to-Hardware_float/apps/hardware_benchmarks/apps/camera_pipeline/bin/process[0x43c37a]
/nobackup/setter/h2h/float_tests/Halide-to-Hardware_float/apps/hardware_benchmarks/apps/camera_pipeline/bin/process[0x435e3c]
/nobackup/setter/h2h/float_tests/Halide-to-Hardware_float/apps/hardware_benchmarks/apps/camera_pipeline/bin/process[0x4355f1]
/nobackup/setter/h2h/float_tests/Halide-to-Hardware_float/apps/hardware_benchmarks/apps/camera_pipeline/bin/process[0x418248]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7ffff5867830]
/nobackup/setter/h2h/float_tests/Halide-to-Hardware_float/apps/hardware_benchmarks/apps/camera_pipeline/bin/process[0x4183f9]
[Inferior 1 (process 26007) exited with code 01]

jeffsetter, 2019-06-26:
After the recent changes in https://github.com/rdaly525/coreir/pull/764, we have the ROM being initialized, but still issues in running:

.........
"63"
AddrBv  = 11111110
Valuebv = 0000000000111111
"63"
AddrBv  = 11111111
Valuebv = 0000000000111111
generated simulated coreir design
reset clock self.clk
reset self.in_arg_0_0_0 with size 16
starting coreir simulation
ERROR: rom__curvea0$mem.raddr cannot be found

../../../../../coreir/lib/libcoreirsim.so(_ZN6CoreIR14SimulatorState9getBitVecEPNS_6SelectE+0x72)[0x7f790c33d8a0]
../../../../../coreir/lib/libcoreirsim.so(_ZN6CoreIR14SimulatorState18updateMemoryOutputEi+0xf8)[0x7f790c344f7e]
../../../../../coreir/lib/libcoreirsim.so(_ZN6CoreIR14SimulatorState16exeCombinationalEv+0x140)[0x7f790c3470dc]
../../../../../coreir/lib/libcoreirsim.so(_ZN6CoreIR14SimulatorState12resetCircuitEv+0x18)[0x7f790c346de4]
./bin/process[0x53494b]
./bin/process[0x418914]
./bin/process[0x437faa]
./bin/process[0x431a6c]
./bin/process[0x431221]
./bin/process[0x417fe3]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f790b186830]
./bin/process[0x418199]
../../hw_support/hardware_targets.mk:146: recipe for target 'run-coreir' failed
make: *** [run-coreir] Error 1

dillonhuff, 2019-06-26:
@jeffsetter so after inlining nothing is connected to the raddr port of the RAM? That seems very odd. Since the error does not show up in the manually wired up unit test.

Are there any other operation types in camera pipeline that are not used in other apps?

Also are you using constant folding before running the simulator? Perhaps some other nodes definition is not wired up correctly and constant folding is removing it.

dillonhuff, 2019-06-26:
@jeffsetter could you send the most recent JSON file for camera pipeline that you used to produce the above error? I am trying to reproduce it in the simulator, but I cannot even get to the line you listed. I get a crash with an exception: "type must be string, but is array"

jeffsetter, 2019-06-26:
Attached is the updated json file. I changed my generation since you made some different assumptions in your code about the ROM JSON.

[rom_design_top.json.txt](https://github.com/rdaly525/coreir/files/3331270/rom_design_top.json.txt)

dillonhuff, 2019-07-26:
@jeffsetter just saw your email. Is the file attached above still current?

jeffsetter, 2019-07-26:
Yes the file is current.

closed 2019-08-22 by jeffsetter