rdaly525/coreir #765: "Modular" compiler output

leonardt, 2019-06-21:
The current coreir backend currently "flattens" the design into a single verilog file. One request we've seen from magma users is that it would be convenient if coreir could generate modules into separate files and also a "listings" file that includes all the generated files. For example, the user might want to have the coreir primitive definitions in a separate verilog file, so the output would be `top.v` and `coreir.v` and a file `top.lst` that contains both those files.

More generally, we could add the capability at the magma level for users to describe how they'd like to generate the output (e.g. mark certain modules to be generated in separate files, or do all modules separate by default and mark certain modules to be packed into a single file).

I suspect this feature would have implications with the linking design, so it would be worth considering whether this would be an orthogonal feature (temporary, stop gap solution) or could be a part of the linking mechanism. For example, we could have the coreir libraries not included by default, and instead the compiler generates the verilog without the definitions, and the user "links" in the coreir library (either via commandline or magma/pycoreir API) which effectively has coreir output the `coreir.v` file and add it to the generated listings file.

I think we were originally doing this approach for simplicity (including a coreir verilog library file), but the user was required to manually manage this (copy the file into the target directory). This proposal would be to have this all managed by coreir and provided by a linking interface, e.g. `coreir -i top.json -o top.v --link coreVerilog coreBitVerilog` (used `--link` because `-l` is currently used by the library loading).

leonardt, 2019-06-21:
Spelunking in the code, looks like there's already a split files option (https://github.com/rdaly525/coreir/blob/bef589737d5918f4e133a113692b40e28ff43262/src/binary/coreir.cpp#L200), so I think the only improvement this would need would be to generate the listings file alongside it. I think splitting files with the listings would be sufficient. For now, if the user wants to pack certain modules together, they could easily do this with `cat` or some other mechanism (note this is much easier then trying to split the modules into separate files)

rdaly525, 2019-06-24:
How much fine grain control do you think users should have over how to split? I think the easiest way is to have an option to split based off of library. This means that each library would be in a single file.

rsetaluri, 2019-06-25:
Just chiming in here: @leonardt `-s` or `--split` will split into multiple files. I can add an option to dump a text file listing the filenames.

leonardt, 2019-06-25:
I think a general per-module mechanism for annotating the desired output file (maybe by default have it be the name of the module) would be great.

For example: it would be nice to generate a listings file with two lists (or two separate listings files): one for the design, one for DV (e.g. behavioral models/stubs), maybe others. This would make it simple and general to pass various files to downstream tools

rsetaluri, 2019-06-25:
Here is a starting point: #766. I think we need more clear definitions for "design" vs. "DV", but that would be great to support that. And I agree someway to annotate the output file per module would be great.

leonardt, 2019-06-25:
I don't think coreir would be figuring out design/dv, this is just an annotation the user would provide.

leonardt, 2019-06-25:
I think more generally it's a way to annotate modules as "groups" in the output.