rdaly525/coreir #767: Specifying simulation functions in CoreIR

rdaly525, 2019-07-08:
I think this would work best if we treated ModuleDef (and/or GeneratorDef) as an Abstract Base Class, then have the simulator code be a concrete class that inherited from that.

@dillonhuff, let me know what you think of this option. We could then move the simulation code for each CoreIR primitive into its own class.

dillonhuff, 2019-07-09:
@rdaly525 why would ModuleDef need to be an abstract class? What would be the downside of just adding a few virtual methods for simulation to the existing Module class?

dillonhuff, 2019-07-09:
Also in order to add black-box plugins we are probably going to have to get rid of the simulators saved state logging and reply.

Is anyone actually using that anymore?

dillonhuff, 2019-07-11:
Also @rdaly525 do you mind if I delete the compiled code simulator tests? That simulator is deprecated anyway and they slow the testbench down a lot.

dillonhuff, 2019-07-14:
Added in #769

dillonhuff, 2019-07-28:
@rdaly525 can we close this issue?

closed 2019-07-31 by rdaly525