rdaly525/coreir #788: Can we remove the Cpp backend tests?

dillonhuff, 2019-08-24:
@rdaly525 @jeffsetter  I think we agreed last year that we were going to deprecate the c++ backend, but the tests never got removed. They take a long time and generate a bunch of output. Is anyone still using the C++ simulator for coreir?

jeffsetter, 2019-08-25:
I am still using the C++ simulator. If there is a better alternative, I can investigate that.

dillonhuff, 2019-08-25:
@jeffsetter thanks for letting me know. I'll leave the tests in.

jeffsetter, 2019-08-25:
I'm still interested in the comment that:
> we agreed last year that we were going to deprecate the c++ backend

Was the thinking that Fault would replace all testing? Perhaps someone could fill me in on what I should be using for CoreIR testing.

dillonhuff, 2019-08-25:
I think the thinking was that we would replace the CoreIR C++ backend with just compiling to Verilog and running Verilator. I didn't realize that anyone was still using the C++ backend.

jeffsetter, 2019-08-25:
Hm, I've had issues simulating the CoreIR implementation of the "rowbuffer" in Verilator. Furthermore, functional models that don't have CoreIR definitions will be difficult to simulate purely in Verilator. Maybe this warrants a further discussion, because I have been thinking myself that using the current CoreIR simulator isn't a long-term solution. And it sounds like it's already been decided before that it will be deprecated.