rdaly525/coreir #791: Support Interpreter Simulation Plugin in PycoreIR

joyliu37, 2019-09-06:
Dillon has helped us create a simulation plugin in the coreIR interpreter to simulate the unified buffer functional model with PE and other logic. Is that possible to have pycoreir supported these functional simulation? 

Currently I am using python to do the rewrite of our unified buffer for CGRA memory. I want to run the functional simulation at coreIR level. Since unified buffer do not have a verilog implemented, I wrote a functional model and plug it in with all the other logic. In order to do the coreIR simulation just as what David do for Aetherling, I want this simulation plugin could be possible at pycoreIR.

joyliu37, 2019-09-06:
Any suggestion is welcome. @dillonhuff @David-Durst @leonardt

rdaly525, 2019-09-11:
This should definitely be possible to do. I suspect the following should work:

1) Compile CoreIR interpreter and link in your Unified Buffer simulator implementation (into the interpreter dynamic library)
2) Install pycoreir from master (not from pip install)

Let me know if this doesn't work