rdaly525/coreir #800: NYI error flattening types on declaration

leonardt, 2019-10-07:
Steps to reproduce:
```
$ cd magma
$ git checkout flatten-types-decl-coreir-bug
$ pytest tests/test_type/test_tuple.py -k test_nested
ERROR: NYI flatten types of generator or nodef module
{global.base_module}.baseIO Is not a flattened type!
  Type is: {'in0':BitIn, 'out0':Bit}
```

The issue is that circuit has a nested type that needs to be flattened but it doesn't have a definition. @rdaly525 I can work on fixing this, do you happen to know if there were any issues related to this that I should look out for?

leonardt, 2019-10-07:
@rdaly525 this seems to fix the above test, I don't think it should affect existing code since it was an error before: https://github.com/rdaly525/coreir/pull/801

Basically, if the module doesn't have a definition, update the port list but don't modify the definition.  I think the problem is that this assumes that the linked module or generator will have the types flattened and there's not a way that this information is stored.

rdaly525, 2019-10-07:
I do not think this ever came up as an issue, so there is nothing tracking it.

The types of a no-definition module or generator should never be modified. This would break the black-box module abstraction. Maybe in the future I can support something like a log of modifications that need to occur on a black box module.

I think what should happen is that flattentypes and verifyflattentypes should be modified to have an optional parameter that specifies whether arbitrary nested arrays of bits are okay (opposed to just a single array of bits).

I would say make this the default behavior, but the coreir interpreter would break, so I think an optional parameter would be better. 

The verilog backend should be updated too.

leonardt, 2019-10-07:
Hmm, so you're saying there will be no way to compile a circuit that refers to a black box module with non-flattened types?

leonardt, 2019-10-07:
In this case, this has a non-flattened tuple type, not an array type.

leonardt, 2019-10-07:
For now, could we just issue a warning saying: black-box module is being modified, assumes that it will be linked with the appropriate type and add an issue for the more general solution later?

The main use case right now is that someone is compiling some magma code with declarations with tuples. They want to fill in the definitions later, and right now there's no way to compile it, so as long as the definition is eventually generated by magma, the modified interface should be compliant.

rdaly525, 2019-10-07:
It depends on what backend you are compiling to. If you are creating verilog, then no you cannot compile a black-box circuit with non-flattened types since verilog does not support non-flattened types. 

(With the exception of nested arrays which could be updated)

leonardt, 2019-10-07:
I think the issue here is the black box module is coming from magma, not verilog, so it does support these non-flattened types.

rdaly525, 2019-10-07:
Sorry, I might be misunderstanding. Is this the flow you want to support?

-Declare circuit in magma (with complex type)
-flatten types of the circuit
-serialize to json
-reload from json
-add a definition for the circuit which compiles with the flattened types

leonardt, 2019-10-07:
They aren't doing any complex linking at the moment.

rdaly525, 2019-10-07:
Got it. So the weird thing is that the user needs to assume exactly how the types are being flattened when implementing the verilog module. I am okay with adding an "unsafe" verilog compilation option that does what you want. This should definitely not be the default behavior though.

rdaly525, 2019-10-07:
Okay, looking into the code more, adding an extra parameter like that would require a lot of changes. If this needs to be fixed ASAP, I am okay for now issuing a warning, and adding a task to update this to an optional unsafe mode.

leonardt, 2019-10-07:
great thanks, I think we can keep this issue open to track it (we can change the title to be more specific), i'll update the code to issue a warning