rdaly525/coreir #815: double free

hofstee, 2019-10-29:
```python
import magma as m

@m.circuit.sequential(async_reset=True)
class Counter1:
    def __init__(self):
        self.count : m.UInt[16] = m.uint(0, 16)

    def __call__(self, inc : m.Bit) -> m.UInt[16]:
        print(locals())
        plusone = self.count + 1
        if inc:
            self.count = plusone
        else:
            self.count = self.count

        O = self.count
        return O

m.compile("counter1", Counter1, output="coreir-verilog")
```

Results in a double-free in coreir.

```
/pycoreir/coreir-cpp/src/binary/coreir.cpp:188 Running Runningvpasses
/pycoreir/coreir-cpp/src/passes/transform/rungenerators.cpp:10 In Run Generators
/pycoreir/coreir-cpp/src/passes/transform/rungenerators.cpp:26 Done running generators
/pycoreir/coreir-cpp/src/binary/coreir.cpp:197 Running vpasses
/pycoreir/coreir-cpp/src/binary/coreir.cpp:238 Modified?: Yes
double free or corruption (fasttop)
```

hofstee, 2019-10-29:
Hmm this might be a pycoreir issue. Closing for now, will file a new issue when I get a better repro.

closed 2019-10-29 by hofstee

leonardt, 2019-10-29:
Can you check if it's happening when you convert the coreir json to verilog? If not, that might suggest it's an issue with the magma pycoreir backend

leonardt, 2019-10-29:
Also could check if it happens when just compiling to coreir (not coreir-verilog)

rdaly525, 2019-10-29:
I recall having issues with double freeing when using pycoreir. Definitely reopen this as a CoreIR issue if you can reproduce it from the coreir binary.

hofstee, 2019-10-29:
@leonardt does not happen with "coreir", does happen with "coreir-verilog".