rdaly525/coreir #853: Inline produces bad verilog

rsetaluri, 2020-03-17:
with the following magma:
```python
import magma as m

class Main(m.Circuit):
    io = m.IO(I0=m.In(m.UInt[8]),
              I1=m.In(m.UInt[8]),
              O=m.Out(m.Bit))
    io.O @= (io.I0 + io.I1)[0]

m.compile("Main", Main, output="coreir-verilog", inline=True)
```

I get the following verilog

```verilog
module Main (
    input [7:0] I0,
    input [7:0] I1,
    output O
);
assign O = magma_Bits_8_add_inst0_out[0];
endmodule
```

Note that if I remove `inline=True`, I get *correct* verilog.

leonardt, 2020-03-17:
I suspect that this is because the assign inline visitor is dispatching on expression nodes, but the type of the index value is a variant (not a general expression).  If this is the case, it may be indirectly fixed by this PR https://github.com/leonardt/verilogAST-cpp/pull/42 which changes the index type to be expressions (in fact, this will be necessary for inlining since the value of the index will now be an expression, not an ID).  I will see if that branch fixes it.

rsetaluri, 2020-03-17:
See my note there...not sure that is valid verilog. Even if it is, ideally inlining should be flexible/modular enough to inline various segments of the expression independently, i.e. if there is something like

```
A(B(x, C(D(y, z))))
```

and B is not "inlineable" we should get something like

```
tmp0 = B(x, C(D(y, z)))
out = A(tmp0)
```

not sure if there's any case outside of bit-selects but it seems like good logic to have in general

leonardt, 2020-03-17:
Yes we can implement that, we'll have to do that to get the desired behavior anyways.

The issue is that the inline logic discovers an inlineable identifier inside the index node, but instead it should stop when it sees an index and assume it's contents can't be inlined

closed 2020-03-18 by leonardt