rdaly525/coreir #863: BFloat round-to-even

jeffsetter, 2020-04-03:
Right now we have floating point +, -, * are working, but instead of round-to-even, it appears that truncation is happening. We should instead be using round-to-even or a flag to switch between round-to-even and truncation.

For example:
7 * 3.140625 (0x4049) = 0x41afe
   Halide BFloat emulated value: 22.0  (0x41b0)
   CoreIR interpreter value: 21.875   (0x41af)

Related:
https://github.com/rdaly525/coreir/issues/727

rdaly525, 2020-04-03:
The code that is causing the issue: https://github.com/rdaly525/coreir/blob/056a6e86770271722baeeddcc6365edf5fe36509/src/simulator/interpret.cpp#L10

cdonovick, 2020-04-03:
According to hwtype implementation the result should be 22.0.

rdaly525, 2020-04-03:
@jeffsetter or @cdonovick, could you reference an appropriate implementation of the round function so I can copy it?

cdonovick, 2020-04-03:
I think you need to round similar https://github.com/StanfordAHA/Halide-to-Hardware/blob/master/src/EmulateFloat16Math.cpp#L220

Although I think that is missing a couple of cases.

After adding half I think there is 4 cases.  .
Cases:
   round up to infinity (if adding half changed the exponent to be full, clear mantissa bits)
   round up incrementing exponent, (if adding half changed exponent bits shift mantissa bits down)
   tie -- round to even (clear bottom bit)
   round down (truncate)

The halide implementation covers the last 2

You also need to check if nan before doing anything and if so make sure you return a NAN value.  I think we can ignore the difference between qNAN and sNAN.

rdaly525, 2020-04-12:
Fixed by #866

closed 2020-04-12 by rdaly525