rdaly525/coreir #896: travis memory leak issue when running tests together

rdaly525, 2020-06-08:
From https://github.com/StanfordAHA/CGRAMapper/issues/91#event-3389197209

I'm not exactly sure what the problem in CoreIR is, but due to a memory leak issue, many tests in the garnet repo cannot be run together with pytest in travis. The tests fail with the message "OSError: [Errno 12] Cannot allocate memory". Here is a link to a build where all the tests fail because of this issue: https://travis-ci.com/github/StanfordAHA/garnet/builds/166559370. To bypass this issue, Keyi split up how the tests are run in the garnet repo in the run.sh script for travis. However, as we add more tests, we keep running into this, so tests have been moved out of test_interconnect to minimize the memory leak but we'd like to move them back in there to test the memory tile with the interconnect if possible. Thanks! (Also not completely sure who to assign to the issue, please feel free to change).

leonardt, 2020-06-08:
Hmm, I wonder if @kavyasreedhar can give us any hints as to why this might be occurring.  Is there anything specific about those tests that might be related to the memory leak? E.g. a large amount of instances or big types or connections?

If I understand correctly, the issue seems to happen when running a lot of tests in the same process, but splitting them up into multiple processes avoids the problem? (e.g. using a script to run different sets of tests)  This might suggest that coreir is not freeing memory across test runs (e.g. every time we compile, we are still holding memory from the previous compile).  I think there's quite a lot of logic that does this (e.g. the context in coreir holds a lot of memory that's only free'd on exit).  If this is the likely cause, I think it would be worth putting some effort into refactoring the coreir memory management patterns that use the context to be smarted about allocating and freeing memory.  This would also likely improve performance by reducing the amount of time spent in the memory management subsystem (e.g. for large designs, we may be unnecessarily requesting more pages of memory when we could be reusing a lot of space)

kavyasreedhar, 2020-06-08:
Lenny, I think your post accurately summarizes what's going on. 

I believe Keyi said the interconnect uses a lot more memory than the memtile which is why if I leave all the tests with the interconnect, then we have to split up the tests, exactly like you said. The workarounds are to split up the tests or move the tests to test just the memory tile, but ideally we would be able to test functionality with the interconnect as well. Thanks!

rdaly525, 2020-06-08:
One quick fix might be to just reset the CoreIR context on each test.

leonardt, 2020-08-12:
See https://github.com/StanfordAHA/garnet/pull/638

leonardt, 2020-08-13:
Memory leak seems to be fixed on garnet master, please feel free to reopen if the issue comes up again.

closed 2020-08-13 by leonardt

kavyasreedhar, 2020-08-13:
Thanks! I'll try moving our test_memory_core tests into test_interconnect and merge all the interconnect tests and let you know.

leonardt, 2020-08-14:
Just FYI, in the latest .travis.yml https://github.com/StanfordAHA/garnet/blob/master/.travis/run.sh all the tests are being run together (so you can merge the files if you'd like and it should still work)

kavyasreedhar, 2020-08-15:
Awesome, thanks!