rdaly525/coreir #902: Trouble running garnet pytest from clean build

mbstrange2, 2020-06-24:
Fresh install of garnet master + 3.7 venv - trying to run `pytest tests/test_interconnect/`

ImportError while loading conftest '/Users/max/Documents/garnet/conftest.py'.
conftest.py:2: in <module>
    from magma import clear_cachedFunctions
venv/src/magma-lang/magma/__init__.py:62: in <module>
    from .compile import *
venv/src/magma-lang/magma/compile.py:3: in <module>
    from .backend import verilog, blif, firrtl, dot, coreir_compiler
venv/src/magma-lang/magma/backend/coreir_compiler.py:3: in <module>
    from .coreir_ import InsertWrapCasts
venv/src/magma-lang/magma/backend/coreir_.py:1: in <module>
    from coreir import Context
venv/lib/python3.7/site-packages/coreir/__init__.py:63: in <module>
    libcoreir_c.COREBitInOut.argtypes = [COREContext_p]
/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ctypes/__init__.py:369: in __getattr__
    func = self.__getitem__(name)
/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ctypes/__init__.py:374: in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
E   AttributeError: dlsym(0x7f8053b27270, COREBitInOut): symbol not found

leonardt, 2020-06-24:
Do you have an old version of coreir installed in your path, perhaps in /usr/local/lib ? You can check with `ls /usr/local/lib | grep coreir`, if so, you'll want to go to wherever you built/installed coreir (probably a local clone of the repo) and run make uninstall to remove the files.  My suspicion is that the pycoreir package is trying to use your local installation rather than the packaged wheel and it's finding an incompatible library (e.g. one compiled before COREBitInOut was introduced).  Alternatively, you can try compiling/installing a new version of coreir from source (this is separate from the python package), but if you don't need bleeding edge features off of master, I would suggest using the pypi package wheel distribution if possible.

rdaly525, 2020-06-24:
From my chat with @mbstrange2:

Okay, I would do the following: 
1) pip uninstall coreir
2) find and remove all coreir/pycoreir files 
  a) deactivate your virtual env, then do another pip uninstall coreir
  b) which coreir to find the installation,
  c) rm path/bin/coreir path/lib/libcoreir*, path/include/coreir.h
3) pip install coreir

That seemed to work, so closing.

closed 2020-06-24 by rdaly525