rdaly525/coreir #908: implicit clock wiring issue?

splhack, 2020-06-26:
magma tests/test_coreir/test_clock_tuple_compile.py fails with master.

      Def:
        Instances:
          Foo_inst0 : Foo
        Connections:
          Foo_inst0.O <=> self.O
          Foo_inst0.I <=> self.I

    ERROR: {Main}.Foo_inst0.clocks_clk Is not fully connected (N)
    {Main}.Foo_inst0.clocks_rst Is not connected

workaround

    +        foo.clocks @= io.clocks

leonardt, 2020-06-26:
Hmm, I'm having trouble reproducing this, can you provide the git SHA for coreir/magma that you're using? (Or pypi version for magma if you're using a release)

splhack, 2020-06-26:
yep,
- coreir https://github.com/rdaly525/coreir/commit/9642814dd847dc82ff99c1c83670e6087dcd934a
- pycoreir https://github.com/leonardt/pycoreir/commit/5c03b6f01cc2f32f2a5f4377e13deb3ecebf520e
- magma https://github.com/phanrahan/magma/commit/ad1b5f791dc8746676913861e3fd29d6317c25aa

With coreir local build

    (cd coreir; cmake; make -j; sudo make install)
    pip3 uninstall coreir
    (cd pycoreir; pip3 install -e .)
    (cd magma; pytest tests/test_coreir/test_clock_tuple_compile.py) # the error

Without local coreir build, but local pycoreir

    (cd coreir; sudo make uninstall)
    pip3 uninstall coreir
    (cd pycoreir; pip3 install -e .)
    (cd magma; pytest tests/test_coreir/test_clock_tuple_compile.py) # pass

With pip coreir

    pip3 uninstall coreir
    pip3 install coreir
    (cd magma; pytest tests/test_coreir/test_clock_tuple_compile.py) # pass

splhack, 2020-06-26:
Hmm, I'm not sure what is going on..
trying to git bisect but even one month old v0.1.38 (https://github.com/rdaly525/coreir/commit/bf49c568333b48d3e61add500a1d6b646dee4859 ) does repros the error.. which means local build has something wrong?

leonardt, 2020-06-26:
hmm that's so strange, I believe the local pycoreir package should be installing from source master so that might suggest that something old is in your build?

you could try making a fresh build/ dir in coreir and installing from there to see if it's something inside your source dir,
```
cd coreir
mkdir build
cd build
cmake ..
sudo make install
```

Can you see if the issue still persists with that?

rdaly525, 2020-06-26:
It also might be the case that you have LD_LIBRARY_PATH pointing to an old CoreIR build.

Lenny's suggestion should work though.

splhack, 2020-06-27:
ok, probably figured out the cause.
it is because `-DSTATIC=ON`. Linux build requires the flag?

I didn't specify it for local build -> error
but pycoreir build did -> no issues

anyway I found a way to install pycoreir with modified coreir 🙂 

    cd coreir
    (modify code)
    cd ../pycoreir
    rm -rf coreir-cpp
    ln -s ../coreir coreir-cpp
    pip3 install -e .

closed 2020-06-27 by splhack