rdaly525/coreir #910: Double free in magma tests

leonardt, 2020-06-29:
See https://travis-ci.org/github/phanrahan/magma/builds/703286416 for an example, it looks like it's in the tests that use commonlib.muxn, so likely related to the recent muxn -> if changes in the verilog backend.  investigating.

leonardt, 2020-06-29:
Interestingly enough, I'm unable to reproduce this issue in a fresh ubuntu 20.04 VM

leonardt, 2020-06-29:
Hmm, nothing on 18.04 either

leonardt, 2020-06-29:
Confirmed it happens on 16.04 but not newer versions of ubuntu

rdaly525, 2020-06-29:
Well that is not a fun bug. I think the addition of a valgrind memory leak test into continuous integration could prevent these kinds of issues in the future

leonardt, 2020-06-29:
Hmm, but when I build coreir from source in the 16.04 VM, the issue goes away, so maybe some compatibility issue with the manylinux distribution and 16.04?

leonardt, 2020-06-30:
Ah, I figured out that the issue arises when I do a static build from source in 16.04, so that helps a bit, will see if having a source build can help debug it.

leonardt, 2020-06-30:
Here's the valgrind log, nothing super useful.  It looks like it's related to the dl code, so perhaps it's https://github.com/rdaly525/coreir/pull/844 and https://github.com/rdaly525/coreir/issues/842 popping up again? I'll try rebasing that branch on master and see what happens.

closed 2020-06-30 by leonardt