rdaly525/coreir #916: Issue with cmake git version in worktree

rsetaluri, 2020-07-08:
Just running vanilla cmake command inside git worktree gives the following error:

```
CMake Warning (dev) at CMakeLists.txt:3 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    CMAKE_PROJECT_VERSION
    CMAKE_PROJECT_VERSION_MAJOR
    CMAKE_PROJECT_VERSION_MINOR
    CMAKE_PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at cmake_modules/GetGitRevisionDescription.cmake:70 (file):
  file failed to open for reading (No such file or directory):

    /Users/rajsekhar/dev/coreir-public/coreir-master/Users/rajsekhar/dev/coreir-public/coreir-dev/.git/worktrees/coreir-master
Call Stack (most recent call first):
  CMakeLists.txt:9 (get_git_head_revision)

CMake Error at cmake_modules/GetGitRevisionDescription.cmake:71 (string):
  string sub-command REGEX, mode REPLACE needs at least 6 arguments total to
  command.
Call Stack (most recent call first):
  CMakeLists.txt:9 (get_git_head_revision)

CMake Warning at CMakeLists.txt:20 (message):
  You are using an untested compiler
```

@splhack this has to do with #909?

leonardt, 2020-07-08:
Maybe a cmake version issue, what version are you using?

leonardt, 2020-07-08:
```
❯ cmake --version
cmake version 3.17.3
```
seems to work for me

splhack, 2020-07-08:
Ubuntu 20.04, cmake 3.16.3 works.
I think I need to update cmake_minimum_required.

rsetaluri, 2020-07-08:
I have 3.16.4. @leonardt to clarify the issue only arises when building inside a **git worktree**. In a vanilla git repo, it works.

splhack, 2020-07-08:
got it. let me search a solution like https://github.com/Z3Prover/z3/pull/1228

rsetaluri, 2020-07-08:
@splhack thank you! If it ends up being too hard, we can have an option in cmake to skip the version information. we only really need it for deploy, and not for dev.

leonardt, 2020-07-08:
ah got it

PR #917 (2020-07-09): update GetGitRevisionDescription for git-worktree
fix #916
closed 2020-07-09 by rsetaluri