rdaly525/coreir #935: Unsupported in C: Cell has the same name as variable: 'rom_curvea0$1_ren'

dillonhuff, 2020-08-11:
I have a new error in rom generation even on the rom-fix branch. Now when I do:

```bash
${COREIR_PATH}/bin/coreir --load_libs commonlib --input camera_pipeline.json --output camera_pipeline.v
```

I get:

```bash
cmd: verilator -Wall --cc camera_pipeline.v --exe --build camera_pipeline_verilog_tb.cpp --top-module camera_pipeline -Wno-lint
%Error: camera_pipeline.v:30527:3: Unsupported in C: Cell has the same name as variable: 'rom_curvea0$1_ren'
30527 | ) rom_curvea0$1_ren (
      |   ^~~~~~~~~~~~~~~~~
        camera_pipeline.v:30491:6: ... Location of original declaration
30491 | wire rom_curvea0$1_ren;
      |      ^~~~~~~~~~~~~~~~~
%Error: camera_pipeline.v:30532:3: Unsupported in C: Cell has the same name as variable: 'rom_curvea0_ren'
30532 | ) rom_curvea0_ren (
      |   ^~~~~~~~~~~~~~~
        camera_pipeline.v:30487:6: ... Location of original declaration
30487 | wire rom_curvea0_ren;
      |      ^~~~~~~~~~~~~~~
%Error: camera_pipeline.v:30431:3: Unsupported in C: Cell has the same name as variable: 'rom_curvea0$2_ren'
30431 | ) rom_curvea0$2_ren (
      |   ^~~~~~~~~~~~~~~~~
        camera_pipeline.v:30402:6: ... Location of original declaration
30402 | wire rom_curvea0$2_ren;
      |      ^~~~~~~~~~~~~~~~~
%Error: camera_pipeline.v:30447:3: Unsupported in C: Cell has the same name as variable: 'rom_curvea0$3_ren'
30447 | ) rom_curvea0$3_ren (
      |   ^~~~~~~~~~~~~~~~~
        camera_pipeline.v:30407:6: ... Location of original declaration
30407 | wire rom_curvea0$3_ren;
      |      ^~~~~~~~~~~~~~~~~
%Error: camera_pipeline.v:30346:3: Unsupported in C: Cell has the same name as variable: 'rom_curvea0$4_ren'
30346 | ) rom_curvea0$4_ren (
      |   ^~~~~~~~~~~~~~~~~
        camera_pipeline.v:30317:6: ... Location of original declaration
30317 | wire rom_curvea0$4_ren;
      |      ^~~~~~~~~~~~~~~~~
%Error: camera_pipeline.v:30362:3: Unsupported in C: Cell has the same name as variable: 'rom_curvea0$5_ren'
30362 | ) rom_curvea0$5_ren (
      |   ^~~~~~~~~~~~~~~~~
        camera_pipeline.v:30322:6: ... Location of original declaration
30322 | wire rom_curvea0$5_ren;
      |      ^~~~~~~~~~~~~~~~~
%Error: camera_pipeline.v:30505:8: Found definition of 'rom_curvea0_ren' as a CELL but expected a variable
30505 | assign rom_curvea0_ren = rom_curvea0_ren_out;
      |        ^~~~~~~~~~~~~~~
%Error: camera_pipeline.v:30512:10: Found definition of 'rom_curvea0_ren' as a CELL but expected a variable
30512 |     .ren(rom_curvea0_ren)
      |          ^~~~~~~~~~~~~~~
%Error: camera_pipeline.v:30516:8: Found definition of 'rom_curvea0$1_ren' as a CELL but expected a variable
30516 | assign rom_curvea0$1_ren = rom_curvea0$1_ren_out;
      |        ^~~~~~~~~~~~~~~~~
%Error: camera_pipeline.v:30523:10: Found definition of 'rom_curvea0$1_ren' as a CELL but expected a variable
30523 |     .ren(rom_curvea0$1_ren)
      |          ^~~~~~~~~~~~~~~~~
%Error: camera_pipeline.v:30420:8: Found definition of 'rom_curvea0$2_ren' as a CELL but expected a variable
30420 | assign rom_curvea0$2_ren = rom_curvea0$2_ren_out;
      |        ^~~~~~~~~~~~~~~~~
%Error: camera_pipeline.v:30427:10: Found definition of 'rom_curvea0$2_ren' as a CELL but expected a variable
30427 |     .ren(rom_curvea0$2_ren)
      |          ^~~~~~~~~~~~~~~~~
%Error: camera_pipeline.v:30436:8: Found definition of 'rom_curvea0$3_ren' as a CELL but expected a variable
30436 | assign rom_curvea0$3_ren = rom_curvea0$3_ren_out;
      |        ^~~~~~~~~~~~~~~~~
%Error: camera_pipeline.v:30443:10: Found definition of 'rom_curvea0$3_ren' as a CELL but expected a variable
30443 |     .ren(rom_curvea0$3_ren)
      |          ^~~~~~~~~~~~~~~~~
%Error: camera_pipeline.v:30335:8: Found definition of 'rom_curvea0$4_ren' as a CELL but expected a variable
30335 | assign rom_curvea0$4_ren = rom_curvea0$4_ren_out;
      |        ^~~~~~~~~~~~~~~~~
%Error: camera_pipeline.v:30342:10: Found definition of 'rom_curvea0$4_ren' as a CELL but expected a variable
30342 |     .ren(rom_curvea0$4_ren)
      |          ^~~~~~~~~~~~~~~~~
%Error: camera_pipeline.v:30351:8: Found definition of 'rom_curvea0$5_ren' as a CELL but expected a variable
30351 | assign rom_curvea0$5_ren = rom_curvea0$5_ren_out;
      |        ^~~~~~~~~~~~~~~~~
%Error: camera_pipeline.v:30358:10: Found definition of 'rom_curvea0$5_ren' as a CELL but expected a variable
30358 |     .ren(rom_curvea0$5_ren)
      |          ^~~~~~~~~~~~~~~~~
%Error: Exiting due to 18 error(s)
        ... See the manual and https://verilator.org for more assistance.
```

It seems verilator does not like the naming conventions used in the generated verilog / coreir. I'm not sure who to point to on this, but perhaps @jeffsetter or @rdaly525 could help me out here?

File: https://github.com/dillonhuff/clockwork/blob/master/coreir_compute/defective/camera_pipeline_cell_error.json

dillonhuff, 2020-08-11:
Sorry just to clarify, the command `${COREIR_PATH}/bin/coreir --load_libs commonlib --input camera_pipeline.json --output camera_pipeline.v` works now, but then when I compile with verilator `verilator -Wall --cc camera_pipeline.v --exe --build camera_pipeline_verilog_tb.cpp --top-module camera_pipeline -Wno-lint` I get this error.

rdaly525, 2020-08-11:
@dillonhuff simple workaround is to add '--inline' to the coreir command. Also reduces code size by ~20%

rdaly525, 2020-08-11:
Not quite sure the exact issue, but I imagine there is an aliasing issue when generating verilog.

dillonhuff, 2020-08-11:
@rdaly525 that did the trick! Thanks man

leonardt, 2020-08-11:
Hmm yea I can take a look, seems like the verilog backend may be introducing a wire that's using a name already in use, but glad inline avoids the problem.

leonardt, 2020-08-11:
I'm going to reopen so we can track the problem until it's fixed

leonardt, 2020-08-25:
Just confirmed that this issue has been resolved in master from another change to how we handle input wires.

closed 2020-08-25 by leonardt