rdaly525/coreir #938: ERROR: Got non-json array for mem init

dillonhuff, 2020-08-12:
@rdaly525 I'm trying to compile a new example with a memory here: https://github.com/dillonhuff/clockwork/blob/master/coreir_compute/defective/accumulation.json

But when I generate verilog I get an error:

```bash
dhuff@kiwi:~/clockwork$ ${COREIR_PATH}/bin/coreir --inline --load_libs commonlib --input ./coreir_compute/defective/accumulation.json --output accumulation.v
ERROR: Got non-json array for mem init

/home/dhuff/clockwork/lib/libcoreir.so(_ZN6CoreIR28convert_mem_init_param_valueEPNS_5ValueEi+0x1bd)[0x7f745b85db14]
/home/dhuff/clockwork/lib/libcoreir.so(_ZN6CoreIR6Passes7Verilog17compileModuleBodyEPNS_10RecordTypeEPNS_9ModuleDefEbbRSt3setINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESt4lessISC_ESaISC_EE+0x142d)[0x7f745b867641]
/home/dhuff/clockwork/lib/libcoreir.so(_ZN6CoreIR6Passes7Verilog13compileModuleEPNS_6ModuleE+0xc30)[0x7f745b86a75a]
/home/dhuff/clockwork/lib/libcoreir.so(_ZN6CoreIR6Passes7Verilog22runOnInstanceGraphNodeERNS_17InstanceGraphNodeE+0x25e)[0x7f745b86b4c2]
/home/dhuff/clockwork/lib/libcoreir.so(_ZN6CoreIR11PassManager20runInstanceGraphPassEPNS_4PassE+0x18b)[0x7f745b73c6f7]
/home/dhuff/clockwork/lib/libcoreir.so(_ZN6CoreIR11PassManager7runPassEPNS_4PassERSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS9_EE+0x39c)[0x7f745b73cb26]
/home/dhuff/clockwork/lib/libcoreir.so(_ZN6CoreIR11PassManager3runERSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS7_EES9_+0x5b0)[0x7f745b73dac8]
/home/dhuff/clockwork/lib/libcoreir.so(_ZN6CoreIR7Context9runPassesESt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS7_EES9_+0x7e)[0x7f745b67d3c0]
/home/dhuff/clockwork/coreir/bin/coreir(main+0x38fe)[0x55ac5222af27]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7)[0x7f745a3c5b97]
/home/dhuff/clockwork/coreir/bin/coreir(_start+0x2a)[0x55ac5220fcca]
```

Any idea whats going on?

leonardt, 2020-08-12:
Looks like the same nested "init" field issue discussed here: https://github.com/rdaly525/coreir/issues/932

I think this line https://github.com/dillonhuff/clockwork/blob/master/coreir_compute/defective/accumulation.json#L2422

should be:             `"modargs":{"init":["Json",[0,1,2,3,4,5,6,7,8]]}`

dillonhuff, 2020-08-12:
@leonardt makes sense!

@jeffsetter could you update the accumulation compute units with the new ROM argument format?

closed 2020-08-12 by dillonhuff