rdaly525/coreir #945: Cannot get two bit types to wire together regardless of direction?

dillonhuff, 2020-08-18:
@rdaly525 I'm working on getting some code for the ASPLOS ubuffer paper working and I get the following error while wiring up some coreir modules:

```bash
ERROR: conv1_stencil_ub: Cannot wire together
   delay_sr_U158.rst_n : BitIn
   self.rst_n : Bit
```

When I flip the type of `rst_n` on `self` I get the error:

```bash
ERROR: conv1_stencil_ub: Cannot wire together
  delay_sr_U158.rst_n : BitIn
  self.rst_n : BitIn
```

Am I misunderstanding the meaning of type directions in coreir? I assumed at least one of these combinations must be legal.

rdaly525, 2020-08-18:
@dillonhuff That first one should absolutely be legal. The issue unfortunately looks like there are two distinct CoreIR contexts being created and used (C++ objects representing CoreIR types are constructed in CoreIR context). Can you make sure that only a single context is ever being created in your flow? If you think you are doing this, can I get a quick summary of the code?

rdaly525, 2020-08-18:
It could also easily be related to https://github.com/rdaly525/coreir/issues/929. Are you using the latest CoreIR which has the assertion added in https://github.com/rdaly525/coreir/pull/930

dillonhuff, 2020-08-18:
@rdaly525 it was the double context issue. I don't know what version I'm on.

closed 2020-08-18 by dillonhuff