rdaly525/coreir #959: Having trouble at branch rm-ubuf-master

francisxu21, 2020-09-03:
I'm having trouble while compiling GarnetFlow repo at https://github.com/StanfordAHA/GarnetFlow. The error is like:

`In file included from /GarnetFlow/scripts/coreir/src/../include/coreir/ir/common.h:4:0,
                 from /GarnetFlow/scripts/coreir/src/../include/coreir.h:6,
                 from /GarnetFlow/scripts/coreir/src/../include/coreir/passes/analysis/verilog.h:4,
                 from /GarnetFlow/scripts/coreir/src/passes/analysis/verilog.cpp:2:
/GarnetFlow/scripts/coreir/src/passes/analysis/verilog.cpp: In function 'std::vector<std::variant<std::unique_ptr<verilogAST::StructuralStatement, std::default_delete<verilogAST::StructuralStatement> >, std::unique_ptr<verilogAST::Declaration, std::default_delete<verilogAST::Declaration> > > > CoreIR::compile_module_body(CoreIR::RecordType*, CoreIR::ModuleDef*, bool)':
/GarnetFlow/scripts/coreir/src/passes/analysis/verilog.cpp:779:44: error: base operand of '->' has non-pointer type 'std::variant<std::unique_ptr<verilogAST::Identifier, std::default_delete<verilogAST::Identifier> >, std::unique_ptr<verilogAST::Vector, std::default_delete<verilogAST::Vector> > >'
         ASSERT(instance_parameters[0].first->value == "value",
                                            ^
/GarnetFlow/scripts/coreir/src/../include/coreir/ir/fwd_declare.h:19:11: note: in definition of macro 'ASSERT'
     if (!(C)) { \
           ^
/GarnetFlow/scripts/coreir/src/passes/analysis/verilog.cpp:785:44: error: base operand of '->' has non-pointer type 'std::variant<std::unique_ptr<verilogAST::Identifier, std::default_delete<verilogAST::Identifier> >, std::unique_ptr<verilogAST::Vector, std::default_delete<verilogAST::Vector> > >'
         ASSERT(instance_parameters[0].first->value == "hi",
                                            ^
/GarnetFlow/scripts/coreir/src/../include/coreir/ir/fwd_declare.h:19:11: note: in definition of macro 'ASSERT'
     if (!(C)) { \
           ^
/GarnetFlow/scripts/coreir/src/passes/analysis/verilog.cpp:787:44: error: base operand of '->' has non-pointer type 'std::variant<std::unique_ptr<verilogAST::Identifier, std::default_delete<verilogAST::Identifier> >, std::unique_ptr<verilogAST::Vector, std::default_delete<verilogAST::Vector> > >'
         ASSERT(instance_parameters[1].first->value == "lo",
                                            ^
/GarnetFlow/scripts/coreir/src/../include/coreir/ir/fwd_declare.h:19:11: note: in definition of macro 'ASSERT'
     if (!(C)) { \
           ^
make[2]: *** [src/CMakeFiles/coreir.dir/build.make:739: src/CMakeFiles/coreir.dir/passes/analysis/verilog.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:432: src/CMakeFiles/coreir.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
`

My tools setting:

-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
Found PythonInterp: /usr/bin/python (found version "3.8.2")

I'm at branch rm-ubuf-master by the way.

rdaly525, 2020-09-03:
@francisxu21 my impression was that GarnetFlow was mostly deprecated in favor of https://github.com/StanfordAHA/aha which should be using the latest CoreIR. is that not the case?

francisxu21, 2020-09-04:
> @francisxu21 my impression was that GarnetFlow was mostly deprecated in favor of https://github.com/StanfordAHA/aha which should be using the latest CoreIR. is that not the case?

It seems like its front end tool, halide-to-hardware still uses this branch. The bash file is like:
`#!/bin/bash -e

SOURCE="${BASH_SOURCE[0]}"
REQUIREMENTS=`(dirname ${SOURCE})`/requirements.txt

# If coreir has not already been built
if [ ! -d "coreir" ]; then
        # test of push ability
        # build coreir here
        #git clone --depth 1 --branch rm-ubuf https://github.com/rdaly525/coreir coreir
        git clone --depth 1 --branch rm-ubuf-master https://github.com/rdaly525/coreir coreir
        # there is a bug in ABI with regex
        # see https://github.com/rdaly525/coreir/issues/737
        #cd coreir && \
        #git apply /GarnetFlow/patches/coreir_path_cxxopt && \
        #cd ..
        cd coreir/build && cmake .. && make -j2 && cd ../../
else
        echo "Coreir already built"
fi`