rdaly525/coreir #972: Why is IEEE compliance for CW floating point add/mul set to 0?

jack-melchert, 2020-10-26:
https://github.com/rdaly525/coreir/blob/367774c87737ad4e09e4cf62c5c7ac6b28825b69/src/libs/float_CW.cpp#L137
https://github.com/rdaly525/coreir/blob/367774c87737ad4e09e4cf62c5c7ac6b28825b69/src/libs/float_CW.cpp#L157

This causes errors when simulating lassen with CW floats:
https://github.com/StanfordAHA/lassen/issues/178

Setting ieee_compliance to true fixes this issue. Is there any reason why it was set to false in the first place?

rdaly525, 2020-10-26:
ieee_compliance is essentially a switch indicating compliance with subnormal/denormal floating point values. If you are seeing testing errors, that might mean that we are generating subnormal numbers in the tests. If I recall correctly, we were not trying to support these. Ill try to dig up an email discussing this issue.

rdaly525, 2020-10-26:
That being said, CoreIR should probably expose that flag as a parameter to be decided by the user of the library