rdaly525/coreir #1009: Compilation error on Apple M1 chip

shi27feng, 2021-05-09:
Hi all,
Does anyone know how can I solve the errors in this compilation? 
Thanks a lot in advance

I quote, "/Users/fengshi/Workspace/coreir/src/../include/coreir/ir/json.h:1636:11: **error**: '**construct**' is deprecated [-Werror,-Wdeprecated-declarations]
    alloc.construct(object.get(), std::forward<Args>(args)...);
          ^"

```shell
❯ cd build
❯ cmake ..
CMake Warning (dev) at CMakeLists.txt:3 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    CMAKE_PROJECT_VERSION
    CMAKE_PROJECT_VERSION_MAJOR
    CMAKE_PROJECT_VERSION_MINOR
    CMAKE_PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning at CMakeLists.txt:33 (message):
  You are using an untested compiler

-- Configuring done
-- Generating done
-- Build files have been written to: /Users/fengshi/Workspace/coreir/build/verilogAST-download
[ 11%] Creating directories for 'verilogAST'
[ 22%] Performing download step (git clone) for 'verilogAST'
-- Avoiding repeated git clone, stamp file is up to date: '/Users/fengshi/Workspace/coreir/build/verilogAST-download/verilogAST-prefix/src/verilogAST-stamp/verilogAST-gitclone-lastrun.txt'
[ 33%] Performing update step for 'verilogAST'
[ 44%] No patch step for 'verilogAST'
[ 55%] No configure step for 'verilogAST'
[ 66%] No build step for 'verilogAST'
[ 77%] No install step for 'verilogAST'
[ 88%] No test step for 'verilogAST'
[100%] Completed 'verilogAST'
[100%] Built target verilogAST
-- /Users/fengshi/Workspace/coreir/src
CMake Deprecation Warning at CMakeLists.txt:3 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done
-- Generating done
-- Build files have been written to: /Users/fengshi/Workspace/coreir/build/tests/gtest/gtest-download
[ 11%] Creating directories for 'googletest'
[ 22%] Performing download step (git clone) for 'googletest'
-- Avoiding repeated git clone, stamp file is up to date: '/Users/fengshi/Workspace/coreir/build/tests/gtest/gtest-download/googletest-prefix/src/googletest-stamp/googletest-gitclone-lastrun.txt'
[ 33%] Performing update step for 'googletest'
HEAD is now at f5e592d8 Googletest export
[ 44%] No patch step for 'googletest'
[ 55%] No configure step for 'googletest'
[ 66%] No build step for 'googletest'
[ 77%] No install step for 'googletest'
[ 88%] No test step for 'googletest'
[100%] Completed 'googletest'
[100%] Built target googletest
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/fengshi/Workspace/coreir/build
❯ make -j4
Consolidate compiler generated dependencies of target gtest
Consolidate compiler generated dependencies of target verilogAST
[... 1115 lines cut ...]
    vjson["interface"] = {
                         ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/usr/include/c++/v1/memory:1682:9: note: 'construct' has been explicitly marked deprecated here
        construct(_Up* __p, _Args&&... __args)
        ^
In file included from /Users/fengshi/Workspace/coreir/src/ir/context.cpp:2:
In file included from /Users/fengshi/Workspace/coreir/src/../include/coreir/ir/context.h:4:
In file included from /Users/fengshi/Workspace/coreir/src/../include/coreir/ir/fwd_declare.h:53:
/Users/fengshi/Workspace/coreir/src/../include/coreir/ir/json.h:1636:11: error: 'construct<std::__1::vector<nlohmann::basic_json<>>, std::initializer_list<nlohmann::basic_json<>> &>' is deprecated [-Werror,-Wdeprecated-declarations]
    alloc.construct(object.get(), std::forward<Args>(args)...);
          ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/usr/include/c++/v1/memory:1680:9: note: 'construct<std::__1::vector<nlohmann::basic_json<>>, std::initializer_list<nlohmann::basic_json<>> &>' has been explicitly marked deprecated here
        _LIBCPP_DEPRECATED_IN_CXX17 _LIBCPP_INLINE_VISIBILITY
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/usr/include/c++/v1/__config:1058:39: note: expanded from macro '_LIBCPP_DEPRECATED_IN_CXX17'
#  define _LIBCPP_DEPRECATED_IN_CXX17 _LIBCPP_DEPRECATED
                                      ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/usr/include/c++/v1/__config:1035:48: note: expanded from macro '_LIBCPP_DEPRECATED'
#    define _LIBCPP_DEPRECATED __attribute__ ((deprecated))
                                               ^
17 errors generated.
make[2]: *** [src/CMakeFiles/coreir.dir/build.make:132: src/CMakeFiles/coreir.dir/ir/context.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:417: src/CMakeFiles/coreir.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

rdaly525, 2021-05-10:
Hello, I have not had the chance to test a coreir build on an M1 yet, but it looks like the json.h (which is copied from https://github.com/nlohmann/json) is using deprecated methods. https://stackoverflow.com/questions/39414610/why-are-are-stdallocators-construct-and-destroy-functions-deprecated-in-c17. I will try to resolve #971 today which should probably fix the issue.

rdaly525, 2021-05-10:
@shi27feng could you replace /Users/fengshi/Workspace/coreir/src/../include/coreir/ir/json.h with https://github.com/nlohmann/json/blob/develop/include/nlohmann/json.hpp and see if that fixes your compilation issues? Unfortunately I do not have a way to reproduce the compilation error.

shi27feng, 2021-05-12:
Hi @rdaly525,
Now I got new errors, I think it should be the "#include" issue, should I copy these files into the folder?

#include <nlohmann/adl_serializer.hpp>
#include <nlohmann/byte_container_with_subtype.hpp>
#include <nlohmann/detail/conversions/from_json.hpp>
#include <nlohmann/detail/conversions/to_json.hpp>
#include <nlohmann/detail/exceptions.hpp>
#include <nlohmann/detail/hash.hpp>
#include <nlohmann/detail/input/binary_reader.hpp>
#include <nlohmann/detail/input/input_adapters.hpp>
#include <nlohmann/detail/input/lexer.hpp>
#include <nlohmann/detail/input/parser.hpp>
#include <nlohmann/detail/iterators/internal_iterator.hpp>
#include <nlohmann/detail/iterators/iter_impl.hpp>
#include <nlohmann/detail/iterators/iteration_proxy.hpp>
#include <nlohmann/detail/iterators/json_reverse_iterator.hpp>
#include <nlohmann/detail/iterators/primitive_iterator.hpp>
#include <nlohmann/detail/json_pointer.hpp>
#include <nlohmann/detail/json_ref.hpp>
#include <nlohmann/detail/macro_scope.hpp>
#include <nlohmann/detail/string_escape.hpp>
#include <nlohmann/detail/meta/cpp_future.hpp>
#include <nlohmann/detail/meta/type_traits.hpp>
#include <nlohmann/detail/output/binary_writer.hpp>
#include <nlohmann/detail/output/output_adapters.hpp>
#include <nlohmann/detail/output/serializer.hpp>
#include <nlohmann/detail/value_t.hpp>
#include <nlohmann/json_fwd.hpp>
#include <nlohmann/ordered_map.hpp>

```shell
❯ make -j4
[  0%] Building CXX object tests/gtest/googletest-build/googletest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
[  0%] Building CXX object verilogAST-build/CMakeFiles/verilogAST.dir/src/verilogAST.cpp.o
[  2%] Building CXX object verilogAST-build/CMakeFiles/verilogAST.dir/src/assign_inliner.cpp.o
[  2%] Building CXX object verilogAST-build/CMakeFiles/verilogAST.dir/src/transformer.cpp.o
[  2%] Building CXX object verilogAST-build/CMakeFiles/verilogAST.dir/src/concat_coalescer.cpp.o
[  3%] Building CXX object verilogAST-build/CMakeFiles/verilogAST.dir/src/zext_coalescer.cpp.o
[  4%] Linking CXX static library ../../../../lib/libgtest.a
[  4%] Built target gtest
[  5%] Building CXX object tests/gtest/googletest-build/googletest/CMakeFiles/gtest_main.dir/src/gtest_main.cc.o
[  6%] Linking CXX shared library ../lib/libverilogAST.dylib
[  6%] Built target verilogAST
[  6%] Building CXX object src/CMakeFiles/coreir.dir/ir/common.cpp.o
[  6%] Building CXX object src/CMakeFiles/coreir.dir/common/logging_lite.cpp.o
[  7%] Building CXX object src/CMakeFiles/coreir.dir/ir/args.cpp.o
[  7%] Linking CXX static library ../../../../lib/libgtest_main.a
[  7%] Built target gtest_main
[  8%] Building CXX object src/CMakeFiles/coreir.dir/ir/constructor.cpp.o
[  9%] Building CXX object src/CMakeFiles/coreir.dir/ir/context.cpp.o
In file included from /Users/fengshi/Workspace/coreir/src/ir/args.cpp:1:
In file included from /Users/fengshi/Workspace/coreir/src/../include/coreir/ir/args.h:2:
In file included from /Users/fengshi/Workspace/coreir/src/../include/coreir/ir/fwd_declare.h:53:
/Users/fengshi/Workspace/coreir/src/../include/coreir/ir/json.h:49:10: fatal error: 'nlohmann/adl_serializer.hpp' file not found
#include <nlohmann/adl_serializer.hpp>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/fengshi/Workspace/coreir/src/ir/common.cpp:1:
In file included from /Users/fengshi/Workspace/coreir/src/../include/coreir/ir/common.h:4:
In file included from /Users/fengshi/Workspace/coreir/src/../include/coreir/ir/fwd_declare.h:53:
/Users/fengshi/Workspace/coreir/src/../include/coreir/ir/json.h:49:10: fatal error: 'nlohmann/adl_serializer.hpp' file not found
#include <nlohmann/adl_serializer.hpp>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [src/CMakeFiles/coreir.dir/build.make:90: src/CMakeFiles/coreir.dir/ir/args.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
1 error generated.
make[2]: *** [src/CMakeFiles/coreir.dir/build.make:104: src/CMakeFiles/coreir.dir/ir/common.cpp.o] Error 1
In file included from /Users/fengshi/Workspace/coreir/src/ir/constructor.cpp:3:
In file included from /Users/fengshi/Workspace/coreir/src/../include/coreir/ir/constructor.h:3:
In file included from /Users/fengshi/Workspace/coreir/src/../include/coreir/ir/fwd_declare.h:53:
/Users/fengshi/Workspace/coreir/src/../include/coreir/ir/json.h:49:10: fatal error: 'nlohmann/adl_serializer.hpp' file not found
#include <nlohmann/adl_serializer.hpp>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/fengshi/Workspace/coreir/src/ir/context.cpp:2:
In file included from /Users/fengshi/Workspace/coreir/src/../include/coreir/ir/context.h:4:
In file included from /Users/fengshi/Workspace/coreir/src/../include/coreir/ir/fwd_declare.h:53:
/Users/fengshi/Workspace/coreir/src/../include/coreir/ir/json.h:49:10: fatal error: 'nlohmann/adl_serializer.hpp' file not found
#include <nlohmann/adl_serializer.hpp>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [src/CMakeFiles/coreir.dir/build.make:118: src/CMakeFiles/coreir.dir/ir/constructor.cpp.o] Error 1
1 error generated.
make[2]: *** [src/CMakeFiles/coreir.dir/build.make:132: src/CMakeFiles/coreir.dir/ir/context.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:417: src/CMakeFiles/coreir.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

rdaly525, 2021-05-12:
@shi27feng interesting, I thought that the link to the json.hpp was standalone. Could you try https://github.com/ArthurSonzogni/nlohmann_json_cmake_fetchcontent/blob/master/include/nlohmann/json.hpp instead? This is what my #1010 PR will be using

shi27feng, 2021-05-12:
Hi @rdaly525,
Now the errors became:

```
/Users/fengshi/Workspace/coreir/src/passes/analysis/verilog.cpp:1349:20: error: 'iterator_wrapper' is deprecated: Since 3.1.0; use items() [-Werror,-Wdeprecated-declarations]
             json::iterator_wrapper(inline_verilog["connect_references"])) {
                   ^
/Users/fengshi/Workspace/coreir/src/../include/coreir/ir/json.h:21273:5: note: 'iterator_wrapper' has been explicitly marked deprecated here
    JSON_HEDLEY_DEPRECATED_FOR(3.1.0, items())
    ^
```

The full error message as below,

```shell
❯ make clean
❯ cmake ..
CMake Warning (dev) at CMakeLists.txt:3 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    CMAKE_PROJECT_VERSION
    CMAKE_PROJECT_VERSION_MAJOR
    CMAKE_PROJECT_VERSION_MINOR
    CMAKE_PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning at CMakeLists.txt:33 (message):
  You are using an untested compiler

-- Configuring done
-- Generating done
-- Build files have been written to: /Users/fengshi/Workspace/coreir/build/verilogAST-download
[ 11%] Performing update step for 'verilogAST'
[ 22%] No patch step for 'verilogAST'
[ 33%] No configure step for 'verilogAST'
[ 44%] No build step for 'verilogAST'
[ 55%] No install step for 'verilogAST'
[ 66%] No test step for 'verilogAST'
[ 77%] Completed 'verilogAST'
[100%] Built target verilogAST
-- /Users/fengshi/Workspace/coreir/src
CMake Deprecation Warning at CMakeLists.txt:3 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done
-- Generating done
-- Build files have been written to: /Users/fengshi/Workspace/coreir/build/tests/gtest/gtest-download
[ 11%] Performing update step for 'googletest'
HEAD is now at f5e592d8 Googletest export
[ 22%] No patch step for 'googletest'
[ 33%] No configure step for 'googletest'
[ 44%] No build step for 'googletest'
[ 55%] No install step for 'googletest'
[ 66%] No test step for 'googletest'
[ 77%] Completed 'googletest'
[100%] Built target googletest
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/fengshi/Workspace/coreir/build

❯ make -j4
[  0%] Building CXX object tests/gtest/googletest-build/googletest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
[... 61 lines cut ...]
[ 37%] Building CXX object src/CMakeFiles/coreir.dir/passes/analysis/smvmodule.cpp.o
[ 38%] Building CXX object src/CMakeFiles/coreir.dir/passes/analysis/smvoperators.cpp.o
[ 38%] Building CXX object src/CMakeFiles/coreir.dir/passes/analysis/verifyconnectivity.cpp.o
[ 39%] Building CXX object src/CMakeFiles/coreir.dir/passes/analysis/verifyflatcoreirprims.cpp.o
[ 39%] Building CXX object src/CMakeFiles/coreir.dir/passes/analysis/verifyflattenedtypes.cpp.o
[ 40%] Building CXX object src/CMakeFiles/coreir.dir/passes/analysis/verifyinputconnections.cpp.o
[ 41%] Building CXX object src/CMakeFiles/coreir.dir/passes/analysis/verilog.cpp.o
[ 41%] Building CXX object src/CMakeFiles/coreir.dir/passes/analysis/verilog/inline_utils.cpp.o
[ 42%] Building CXX object src/CMakeFiles/coreir.dir/passes/analysis/vmodule.cpp.o
/Users/fengshi/Workspace/coreir/src/passes/analysis/verilog.cpp:1349:20: error: 'iterator_wrapper' is deprecated: Since 3.1.0; use items() [-Werror,-Wdeprecated-declarations]
             json::iterator_wrapper(inline_verilog["connect_references"])) {
                   ^
/Users/fengshi/Workspace/coreir/src/../include/coreir/ir/json.h:21273:5: note: 'iterator_wrapper' has been explicitly marked deprecated here
    JSON_HEDLEY_DEPRECATED_FOR(3.1.0, items())
    ^
/Users/fengshi/Workspace/coreir/src/../include/coreir/ir/json.h:1055:112: note: expanded from macro 'JSON_HEDLEY_DEPRECATED_FOR'
    #define JSON_HEDLEY_DEPRECATED_FOR(since, replacement) JSON_HEDLEY_DIAGNOSTIC_DISABLE_CPP98_COMPAT_WRAP_([[deprecated("Since " #since "; use " #replacement)]])
                                                                                                               ^
[ 42%] Building CXX object src/CMakeFiles/coreir.dir/passes/transform/add_dummy_inputs.cpp.o
1 error generated.
make[2]: *** [src/CMakeFiles/coreir.dir/build.make:846: src/CMakeFiles/coreir.dir/passes/analysis/verilog.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:417: src/CMakeFiles/coreir.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

rdaly525, 2021-05-12:
Okay I think that is a good error that is fixed by #1010 which I will be trying to merge shortly. Feel free to try out that branch (git checkout json)

shi27feng, 2021-05-12:
Thanks, I'll try later, and thank you for this very useful project.

shi27feng, 2021-05-12:
Hi @rdaly525,
The good news is that the compilation and installation passed without any issue now.
The bad news is that the test code is still not working properly, the error is still related to "json.hpp"

```shell
❯ make -j test
make -C tests
make[1]: Entering directory '/Users/fengshi/Workspace/coreir/tests'
make -C unit
make[2]: Entering directory '/Users/fengshi/Workspace/coreir/tests/unit'
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/add4.o add4.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/addN.o addN.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/bit_in_out.o bit_in_out.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/can_sel.o can_sel.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/casting.o casting.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/change_register_default.o change_register_default.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/clockify_interface.o clockify_interface.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/comb.o comb.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/connect.o connect.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/constant_folding.o constant_folding.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/cull_zexts.o cull_zexts.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/dadd4.o dadd4.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/defaulttest.o defaulttest.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/delete_dead_instances.o delete_dead_instances.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/easy_sequential.o easy_sequential.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/ice40.o ice40.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/inlinetest.o inlinetest.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/jsontype.o jsontype.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/libtest.o libtest.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/logging.o logging.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/moduleparam.o moduleparam.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/pack_bit_constants.o pack_bit_constants.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/pack_connections.o pack_connections.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/printer.o printer.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/rename_yosys_auto_generated_instances.o rename_yosys_auto_generated_instances.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/replace_port.o replace_port.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/serializationUnit.o serializationUnit.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/simple.o simple.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/split_files.o split_files.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/split_inout_port.o split_inout_port.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/streamifyUnit.o streamifyUnit.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/subcircuit.o subcircuit.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/test_markdirty.o test_markdirty.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/typegen_implicit.o typegen_implicit.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/typetest.o typetest.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/valuetest.o valuetest.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -I../../build/_deps/json-src/include/ -c -o build/wire_clock_pass.o wire_clock_pass.cpp

[... cut ...]
../../include/coreir/ir/fwd_declare.h:15:10: fatal error: 'nlohmann/json.hpp' file not found
#include "nlohmann/json.hpp"
         ^~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [Makefile:21: build/add4.o] Error 1
make[2]: *** Waiting for unfinished jobs....
1 error generated.
make[2]: *** [Makefile:21: build/float_CW.o] Error 1
make[2]: Leaving directory '/Users/fengshi/Workspace/coreir/tests/float'
make[1]: *** [Makefile:7: all] Error 2
make[1]: Leaving directory '/Users/fengshi/Workspace/coreir/tests'
make: *** [Makefile:9: test] Error 2
❯
```

rdaly525, 2021-05-12:
Just merged in the json changes to master. Could you check if that fixes your issue or if you are still seeing an error?

shi27feng, 2021-05-13:
Hi @rdaly525,
Here's what I got from testing, is it fine even without passing these tests? I try magma with coreir seems no problem on Macbook Air M1.

```shell
❯ make -j test
make -C tests
make[1]: Entering directory '/Users/fengshi/Workspace/coreir/tests'
make -C unit
make[2]: Entering directory '/Users/fengshi/Workspace/coreir/tests/unit'
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -c -o build/jsontype.o jsontype.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -c -o build/valuetest.o valuetest.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -o build/valuetest build/valuetest.o -L../../lib -Wl,-rpath,../../lib -lverilogAST -lcoreir -lcoreirsim -lcoreir-aetherlinglib -lcoreir-commonlib
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -o build/jsontype build/jsontype.o -L../../lib -Wl,-rpath,../../lib -lverilogAST -lcoreir -lcoreirsim -lcoreir-aetherlinglib -lcoreir-commonlib
rm build/jsontype.o build/valuetest.o
make[2]: Leaving directory '/Users/fengshi/Workspace/coreir/tests/unit'
make -C unit-c
make[2]: Entering directory '/Users/fengshi/Workspace/coreir/tests/unit-c'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/Users/fengshi/Workspace/coreir/tests/unit-c'
make -C ext
make[2]: Entering directory '/Users/fengshi/Workspace/coreir/tests/ext'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/Users/fengshi/Workspace/coreir/tests/ext'
make -C float
make[2]: Entering directory '/Users/fengshi/Workspace/coreir/tests/float'
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -c -o build/add4.o add4.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -c -o build/float_CW.o float_CW.cpp
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -o build/add4 build/add4.o -L../../lib -Wl,-rpath,../../lib -lcoreir-float -lcoreir
g++ -std=c++17 -Wall -fPIC -Werror -I../../include -I. -o build/float_CW build/float_CW.o -L../../lib -Wl,-rpath,../../lib -lcoreir-float -lcoreir
rm build/float_CW.o build/add4.o
make[2]: Leaving directory '/Users/fengshi/Workspace/coreir/tests/float'
make -C simulator
make[2]: Entering directory '/Users/fengshi/Workspace/coreir/tests/simulator'
Makefile:24: warning: overriding recipe for target 'build/simtests'
Makefile:17: warning: ignoring old recipe for target 'build/simtests'
g++ -std=c++17 -Wall -fPIC -Werror -Wno-unused-variable -I../../include -I. -c -o build/aetherlingHelpersSim.o aetherlingHelpersSim.cpp
g++ -std=c++17 -Wall -fPIC -Werror -Wno-unused-variable -I../../include -I. -c -o build/circuit_gen.o circuit_gen.cpp
g++ -std=c++17 -Wall -fPIC -Werror -Wno-unused-variable -I../../include -I. -c -o build/combinational.o combinational.cpp
g++ -std=c++17 -Wall -fPIC -Werror -Wno-unused-variable -I../../include -I. -c -o build/fuzzing.o fuzzing.cpp
g++ -std=c++17 -Wall -fPIC -Werror -Wno-unused-variable -I../../include -I. -c -o build/interpreter.o interpreter.cpp
g++ -std=c++17 -Wall -fPIC -Werror -Wno-unused-variable -I../../include -I. -c -o build/moduleparamsim.o moduleparamsim.cpp
g++ -std=c++17 -Wall -fPIC -Werror -Wno-unused-variable -I../../include -I. -c -o build/partial_evaluation.o partial_evaluation.cpp
g++ -std=c++17 -Wall -fPIC -Werror -Wno-unused-variable -I../../include -I. -c -o build/sequential.o sequential.cpp
g++ -std=c++17 -Wall -fPIC -Werror -Wno-unused-variable -I../../include -I. -c -o build/stress.o stress.cpp
In file included from stress.cpp:3:
./catch.hpp:9007:5: error: unrecognized instruction mnemonic, did you mean: bit, cnt, hint, ins, not?
    CATCH_BREAK_INTO_DEBUGGER();
    ^
./catch.hpp:8704:38: note: expanded from macro 'CATCH_BREAK_INTO_DEBUGGER'
    if (Catch::isDebuggerActive()) { CATCH_TRAP(); }                           \
                                     ^
./catch.hpp:8681:30: note: expanded from macro 'CATCH_TRAP'
#define CATCH_TRAP() __asm__("int $3\n" : :) /* NOLINT */
                             ^
<inline asm>:1:2: note: instantiated into assembly here
        int $3
        ^
1 error generated.
make[2]: *** [Makefile:27: build/stress.o] Error 1
make[2]: Leaving directory '/Users/fengshi/Workspace/coreir/tests/simulator'
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/Users/fengshi/Workspace/coreir/tests'
make: *** [Makefile:9: test] Error 2
```